iNavFlight/inav-configurator #1811: FC-Mamba 405: SBUS-Receivers don't work in INAV-Configurator, but work fine in Betaflight

hermyFri, 2023-07-26:
### **Current Behaviour**

SBUS-Communication problem; No reaction in Tab "Receiver", when Remote Control Sticks are moved

### **Steps to Reproduce**

1. Connect a "FrSky XM+" or "Futaba R2001SB 2.4 GHZ S-FHSS" SBUS Receiver to UART1    
2. Turn On RemoteControl -> Green LED of the Receiver is active
3. Go to Tab "Receiver"; move Remote Control Stick -> Colored channel bars don't move

-> The same hardware configuration works without problems with Betaflight Configurator

### **Expected Behaviour**
moving channel bars on Remote Control commands

### **Suggested Solution**
?

INAV Configurator 6.1.0, FC , Betaflight Configurator 10.9
Firmware 6.1.1 FURYF4OSD

FC Board name and vendor: Diatone Mamba F405 MK2

INAV-Configuration:

[image]
[image]

kotikvadik, 2023-08-13:
Hi, I have the same problem with FC Mamba F405 MK2. Using 6.1.1 FURYF4OSD firmware it is not sensetive to sticks movement. But on BF there's no such a problem. So I am simply using 6.1.1 MambaF405 firmware.

hermyFri, 2023-08-13:
Hi kotikvadik

When I use  6.1.1 MambaF405 firmware, the S-Bus receiver works.
Thanks a lot for your very useful hint !

kotikvadik, 2023-08-13:
You are welcome. But there is one more thing with this firmware. If you use "Throttle Position" and "Scaled Throttle Position" in the OSD menu, they show different data even if the quad not in alt hold mode. In the previous version of the firmware, both showed the same. So be careful. Also I have а problem with baro.

closed 2023-09-29 by DzikuVx