iNavFlight/inav-configurator #1817: SITL Ser2TCP issue

sevrugin, 2023-08-09:
I'm trying to use serial2tcb bridge to connect my ELRS receiver by usr2ttl adapter to SITL
Ubuntu 23.04
I have an error
```
[Serial2TCP] Unable to open /dev/ttyUSB0: System.ArgumentException: Value does not fall within the expected range.
   at System.IO.Ports.SerialStream..ctor(String, Int32, Parity, Int32, StopBits, Int32, Int32, Handshake, Boolean, Boolean, Boolean, Byte) + 0x370
   at System.IO.Ports.SerialPort.Open() + 0xee
   at Ser2TCP.Program.Ser2TCP(String, Int32, Int32, StopBits, Parity, IPAddress, Int32, Int32) + 0x11a
```
If I'm trying to run this command manually - error the same
`./Ser2TCP --comport /dev/ttyUSB0 --baudrate 420000 --tcpport 5762 --ip 0.0.0.0`

I tried to use another program (python serial2tcp) and it works correct

stronnag, 2023-08-09:
We should really replace these binary programs (in particular the Linux monstrosity) with code we own / control.
 May I be as presumptive to propose choosing one of the language variants from these [Cross platform](https://github.com/stronnag/bbl2kml/tree/master/tools/ser2tcp) ser2tcp ser2tcp implementations, preferably the [C language]( https://github.com/stronnag/bbl2kml/tree/master/tools/ser2tcp/eg_c) option.

Note the C-lang instances are what is shipped with the so called  "[Next](http://seyrsnys.myzen.co.uk/inav-configurator-next/) "configurators.  You may find working "drop in" replaceable binaries [in the archives here ](https://github.com/stronnag/bbl2kml/wiki/fl2sitl#images)

sevrugin, 2023-08-09:
Rebooted for a clean result
Ubuntu 23.04
Launched `fl2sitl --minimal`
Launched Inav configurator 7 last
Compiled and copy ser2tcp
Pressed start
[image]

```
$ lsof -p $(pidof inav_SITL)
COMMAND    PID     USER   FD      TYPE             DEVICE SIZE/OFF     NODE NAME
inav_SITL 5013 sevrugin  cwd       DIR                8,2     4096 15500572 /home/sevrugin/Programms/INAV Configurator
inav_SITL 5013 sevrugin  rtd       DIR                8,2     4096        2 /
inav_SITL 5013 sevrugin  txt       REG                8,2   514440 15496830 /home/sevrugin/Programms/INAV Configurator/resources/sitl/linux/inav_SITL
inav_SITL 5013 sevrugin    0u     unix 0x0000000000000000      0t0    39934 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin    1u     unix 0x0000000000000000      0t0    39936 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin    2u     unix 0x0000000000000000      0t0    48130 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin    3u     unix 0x0000000000000000      0t0    44371 @605b4 type=SEQPACKET (CONNECTED)
inav_SITL 5013 sevrugin    4u     unix 0x0000000000000000      0t0    44374 type=SEQPACKET (CONNECTED)
inav_SITL 5013 sevrugin    5u     unix 0x0000000000000000      0t0    42360 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin    6u     IPv4              45397      0t0      UDP *:58417 
inav_SITL 5013 sevrugin    7r      REG                8,2 10269664 15497303 /home/sevrugin/Programms/INAV Configurator/icudtl.dat
inav_SITL 5013 sevrugin    8r      REG                8,2   671752 15496747 /home/sevrugin/Programms/INAV Configurator/v8_context_snapshot.bin
inav_SITL 5013 sevrugin    9u      REG                8,2    32768 18222276 /home/sevrugin/.config/inav-configurator/Default/x_plane_crsf.bin
inav_SITL 5013 sevrugin   10u     IPv6              45402      0t0      TCP *:5760 (LISTEN)
inav_SITL 5013 sevrugin   11u     IPv6              45407      0t0      TCP *:5761 (LISTEN)
inav_SITL 5013 sevrugin   96r      REG                8,2   632688 15496969 /home/sevrugin/Programms/INAV Configurator/nw_100_percent.pak
inav_SITL 5013 sevrugin   97r      REG                8,2   954144 15496968 /home/sevrugin/Programms/INAV Configurator/nw_200_percent.pak
inav_SITL 5013 sevrugin   98r      REG                8,2   298120 15497059 /home/sevrugin/Programms/INAV Configurator/locales/en-US.pak
inav_SITL 5013 sevrugin   99r      REG                8,2  3924612 15496828 /home/sevrugin/Programms/INAV Configurator/resources.pak
inav_SITL 5013 sevrugin  103r      REG                8,2   671752 15496747 /home/sevrugin/Programms/INAV Configurator/v8_context_snapshot.bin
inav_SITL 5013 sevrugin  116u     unix 0x0000000000000000      0t0    44397 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin  117u     unix 0x0000000000000000      0t0    43399 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin  120u      REG               0,25      144       84 /dev/shm/.io.nwjs.9OKO55 (deleted)
inav_SITL 5013 sevrugin  122r      REG               0,22        0    42364 /proc/4813/statm
inav_SITL 5013 sevrugin  123r      REG               0,22        0    45372 /proc/4813/status
inav_SITL 5013 sevrugin  124u      REG                8,2  2379080  9175058 /tmp/.io.nwjs.ZkjjJE (deleted)
inav_SITL 5013 sevrugin  127u      REG               0,25    65536      157 /dev/shm/.io.nwjs.nU6Mb6 (deleted)
inav_SITL 5013 sevrugin  128u  a_inode               0,14        0     2081 [eventfd:66]
inav_SITL 5013 sevrugin  129u      REG               0,25      144       87 /dev/shm/.io.nwjs.cyDkMQ (deleted)
inav_SITL 5013 sevrugin  130u      REG               0,25  2097152      159 /dev/shm/.io.nwjs.51L3ac (deleted)
inav_SITL 5013 sevrugin  131u      REG               0,25    65536       89 /dev/shm/.io.nwjs.w60Mrc (deleted)
inav_SITL 5013 sevrugin  132u      REG               0,25    65536       90 /dev/shm/.io.nwjs.nJXgY5 (deleted)
inav_SITL 5013 sevrugin  133u     unix 0x0000000000000000      0t0    44407 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin  136u  a_inode               0,14        0     2081 [eventfd:64]
inav_SITL 5013 sevrugin  138u      REG               0,25  4198400      123 /dev/shm/.io.nwjs.lXU2fo (deleted)
inav_SITL 5013 sevrugin  139r      REG                8,2   330412  9227757 /usr/share/fonts/truetype/msttcorefonts/Times_New_Roman.ttf
inav_SITL 5013 sevrugin  140u      REG               0,25     4096      125 /dev/shm/.io.nwjs.uSIAjN (deleted)
inav_SITL 5013 sevrugin  141r      REG                8,2   275572  9227744 /usr/share/fonts/truetype/msttcorefonts/Arial.ttf
inav_SITL 5013 sevrugin  142r      REG                8,2   759720  6424283 /usr/share/fonts/truetype/dejavu/DejaVuSans.ttf
inav_SITL 5013 sevrugin  143u      REG               0,25    65536      158 /dev/shm/.io.nwjs.KBlSl0 (deleted)
inav_SITL 5013 sevrugin  144u      REG               0,25     4096      104 /dev/shm/.io.nwjs.0rJC7k (deleted)
inav_SITL 5013 sevrugin  146r      REG                8,2   510612  6423854 /usr/share/fonts/truetype/noto/NotoSansMono-Regular.ttf
inav_SITL 5013 sevrugin  148r      REG                8,2   380660  6424289 /usr/share/fonts/truetype/dejavu/DejaVuSerif.ttf
inav_SITL 5013 sevrugin  149r      REG               0,25  1048576      141 /dev/shm/.io.nwjs.umgjsG (deleted)
inav_SITL 5013 sevrugin  150u     unix 0x0000000000000000      0t0    39866 type=STREAM (CONNECTED)
inav_SITL 5013 sevrugin  153u  a_inode               0,14        0     2081 [eventfd:74]
```

stronnag, 2023-08-09:
Your SITL is not listening on `TCP:5762`
* it is not shown in the Configurator window
* nor is it shown in the `lsof` listing.

This one is:
[image]

```
$ fl2sitl --minimal -listen 49001
```

stronnag, 2023-08-09:
You have configured `UART3` for SerialRX in for `SITL`?

stronnag, 2023-08-09:
I can't see it, you didn't show the Ports tab.

sevrugin, 2023-08-09:
Yohoo!!!
Finally, it is connected (I chose UART3 as a SerialRX on Ports tab, and THEN re-run SITL).
```
INAV 6.1.0 SITL
[SYSTEM] Init...
[SIM] Waiting for connection...
[SOCKET] xplane address = 127.0.0.1:49000, fd=6
[SIM] Connection with X-Plane successfully established.
[EEPROM] Loaded '/home/sevrugin/.config/inav-configurator/Default/x_plane_crsf.bin' (32768 of 32768 bytes)
[SOCKET] Bind TCP :: port 5760 to UART1
[SOCKET] Bind TCP :: port 5761 to UART2
[SOCKET] Bind TCP :: port 5762 to UART3
[SOCKET] ::ffff:127.0.0.1 connected to UART3
[Serial2TCP] Connected to 127.0.0.1:5762 <-> /dev/ttyUSB0
```

Don't see any stick actions, but I'll double check all the configurations and tell you later

sevrugin, 2023-08-10:
Ready to continue )
As I previously said - I don't see any data from receiver - ExpressLRS EP2 (CRSF protocol)
Steps:
The receiver is conneceted by TTL convertor
`screen /dev/ttyUSB0 420000` - works (can see some packets in output)
`nc -l 5372` + `./Ser2TCP -c /dev/ttyUSB0 -b 420000 -i 127.0.0.1 -p 5762` - works (can see some packets in "nc" output)

Trying with InavConfigurator 6.1
`./fl2sitl --minimal`
Inav Ports configured as MSP for UART1 and UART2, and SerialRX for UART3
```
INAV 6.1.0 SITL
[SYSTEM] Init...
[SIM] Waiting for connection...
[SOCKET] xplane address = 127.0.0.1:49000, fd=6
[SIM] Connection with X-Plane successfully established.
[EEPROM] Loaded '/home/sevrugin/.config/inav-configurator/Default/x_plane_crsf.bin' (32768 of 32768 bytes)
[SOCKET] Bind TCP :: port 5760 to UART1
[SOCKET] Bind TCP :: port 5761 to UART2
[SOCKET] Bind TCP :: port 5762 to UART3
[SOCKET] ::ffff:127.0.0.1 connected to UART3
[Serial2TCP] Connected to 127.0.0.1:5762 <-> /dev/ttyUSB0
```
Receiver TAB - Serial + CRSF
Moving stick on TX not showing on the tab

PS: still can connect on /dev/ttyUSB0 and see data (I thought that ttyUSB0 should be locked after the bridge connection)
`telnet 127.0.0.1 5762` successful but doesn't show any data

[image]

[image]

sevrugin, 2023-08-10:
while writing, the data was changed few times
but the data still incorrect
is it possible that buadrate is incorrect?
[image]

stronnag, 2023-08-10:
Works perfectly here (CRSF). It is probable that your USB-TTL adaptor cannot handle "non-standard" baud rates.
For example:
* Ancient "Prolific" `pl2303` device. When set to `420000` baud it actually uses `421052` which is within 0.25% and works find with CRSF.
* Newer `cp210x` device, only supports "POSIX like" baud rates. When set to `420000` baud it actually uses `460800` which differs by 9.7% and I doubt would work with CRSF.

So again, SITL and ser2tcp work perfectly if you've (a) configured everything correctly and (b) you use adequate hardware.

Maybe I'll add a warning to `ser2tcp` where the desired and actual baud rates differ ...

sevrugin, 2023-08-10:
I'm using CH340 TTL adapter
Exactly like in image
[image]
But you are right. This is possible that it doesn't support 420000

stronnag, 2023-08-10:
The only way is to test it, I'm going to update the 'C' ser2tcp to provide a warning. I'll let you know when it's done.

stronnag, 2023-08-10:
You can ungzip /  rename and try the following. It will report any difference in desired baudrate and device actual rate. This seems only  to be a Linux issue; on  FreeBSD, MacOS and Windows the driver reports the speed as requested.

```
$ ser2tcp -c /dev/ttyUSB0 -b 420000 
Warning: device speed 421052 differs from requested 420000
Connected to [::1]:5762 <-> /dev/ttyUSB0
```
This small difference is OK, the converter works just fine with configurator / SITL. The other device that sets itself to 460800 does not work with the SITL / Configurator.
[ser2tcp-static.gz](https://github.com/iNavFlight/inav-configurator/files/12314913/ser2tcp-static.gz)

sevrugin, 2023-08-10:
I'll try your version as well a bit later
How I fixed my problem with ExpressLRS - I reflashed receiver and set baudrate to 400000
[image]
The same speed for ser2tcp
It works now )

RomanLut, 2023-10-12:
https://github.com/iNavFlight/inav/pull/9365

closed 2025-01-18 by b14ckyy