iNavFlight/inav-configurator #1821: Incorrect Output Mapping

shota3527, 2023-09-10:
@mmosca 
Thanks for the efforts on smarter pwm mapping. But on the latest master.
1. Servo is not displayed if platform type is multi-rotor/tri copter
[image]

1. Incorrect output mapping if servo index jumped. In actual mapping, servo index 2,3 is mapped to S2,S3.  servo3 should be on S3 in the GUI.
[image]
```
# resource
IO:
----------------------
...omitted
C06: SERVO1 OUT
C07: SERVO2 OUT
C08: SERVO3 OUT
...omitted
# smix
smix 0 1 2 100 0 -1
smix 1 3 0 100 0 -1

# mmix
```

How about using getMotorCount() and getServoCount() in the firmware to get the correct count as a solution?

mmosca, 2023-09-10:
was this after save and reboot?

mmosca, 2023-09-10:
And what target was it?

Was it on master?
I certainly have seen that bug in development, but though it was fixed.

stronnag, 2023-09-10:
Same UI artefact (master of F/w and configurator).  Set in the CLI. MATEKF405.
[image]

shota3527, 2023-09-10:
> And what target was it?
> 
> Was it on master? I certainly have seen that bug in development, but though it was fixed.

It is matekf405, but I have seen something similar on matekf405wte.
The reboot has no effect.

For Incorrect output mapping if servo index jumped, I think it existed a long time ago

mmosca, 2023-09-10:
let me try to reproduce it here

mmosca, 2023-09-10:
#1822 should fix this.

commit ac1cc38f (2023-09-10): Fix #1821

The code was checking the wrong variable for usage flags, after refactoring to inform configurator of timers
closed 2023-09-10 by mmosca