iNavFlight/inav-configurator #1886: [7.0.0-RC1] GPS Update time statistics a bit jumpy

0crap, 2023-11-05:
Not sure if this is an issue, so please close if normal behavior.

I noticed, on the GPS tab -> Statistics -> Update time: the value is quite jumpy.
On 10Hz (now the default) it jumps between 8 and 12 hz. With an occasional jump as low as 2 Hz.
When I use the CLI to set to 5Hz the stats keep being jumpy. With an occasional jump to as high as 10 Hz.

Tried this on two GPS units.
- Walksnail [WS-M181 GPS](https://caddxfpv.com/products/walksnail-ws-m181gps)  (u-blox M10)
- Beitian BN-220 GPS 

Both configured as `set gps_provider = UBLOX7`
If i need to flash back to INAV v6.1.1, to verify behavior, just let me know.

mmosca, 2023-11-05:
Same thing occurs on 6.1.1 with m8. The way the update rate is calculated is not super strict, and depending on how the gps updates line up with the fc update rates it may fluctuate a bit.

closed 2023-11-05 by mmosca

0crap, 2023-11-05:
Fair enough, thx for your time.