iNavFlight/inav-configurator #1893: Bing Map Provider has massive altitude discrepancies in Elevation Data

b14ckyy, 2023-11-15:
OSM Maplink Elevation Data differ massively from Bing Maps Elevation Data in configurator. 
OSM seems to be the correct one as it is on par with other providers like GMaps, Airmap and Copernicus within a 5m range or less. 
Depending on the region I checked Bing Maps shows +40-45m in Germany. up to +40m in UK. -40m in the US East coast and +20m on Hawaii. 
These errors are too big to be ignored and are not suitable for a ASL planned mission or automatic landing at all. 

Other Applications like MultiWiiPlanner also use Bing maps as a map provider and they show the correct data so I assume this is a parsing error in INAV Configurator. 

[image]
[image]

stronnag, 2023-11-15:
The problem it would appear is that the configurator is either using OpenTopo elevations for OSM maps and Bing  elevations with ellipsoid datum for Bing Imagery (or it's using Bing elevations with sealevel datum  for OSM and Bing elevations with ellipsoid datum for imagery.

For each of your locations (so the point moved ... but it makes no difference; the datum is the issue.
For example:

```
$  ./get_elevations -sea-level -bing-key $(<bkey.txt) 51.4499118 11.8806571
Using mapzen topo set
topo : 51.449912, 11.880657, elev = 119.0 in 346.475603ms
bing (sea-level) : 51.449912, 11.880657, elev = 118.0 in 201.815501ms

$ ./get_elevations -bing-key $(<bkey.txt) 51.4499118 11.8806571
Using mapzen topo set
topo : 51.449912, 11.880657, elev = 119.0 in 127.887764ms
bing (ellipsoid) : 51.449912, 11.880657, elev = 163.0 in 78.228758ms
```

and 

```
./get_elevations -bing-key $(<bkey.txt) 51.4498917 11.8805605
Using mapzen topo set
topo : 51.449892, 11.880560, elev = 119.0 in 122.580514ms
bing (ellipsoid) : 51.449892, 11.880560, elev = 163.0 in 239.017463ms

[jrh@eeyore]$ ./get_elevations -sea-level -bing-key $(<bkey.txt) 51.4498917 11.8805605
Using mapzen topo set
topo : 51.449892, 11.880560, elev = 119.0 in 118.168594ms
bing (sea-level) : 51.449892, 11.880560, elev = 118.0 in 201.112984ms
```

Where the configurator  is using Bing for elevations, it should be using sea-level (as that's what INAV gets from the GPS).

Scavanger, 2023-11-15:
There is even the option to use "sea level" for Bing maps, but for some reason the option is hidden under "Plot Elevation" and is not saved.
The option is actually completely useless and "sea level" should always be used. 
[image]

breadoven, 2023-11-16:
Bing offers an `ellipsoid` or `sealevel `option for elevation. The `sealevel `option can be switched on in the Take Off Home box (`MP elevation` button on the map) by selecting the "Use sea level earth DEM model". I seem to remember the problem with `ellipsoid `vs `sealevel `option is the accuracy seems to vary wildly depending on where you are on the planet. It's not always the case that `sealevel `is the more accurate so you need to chose the one that best fits your local area. However, I guess if `sealevel ` generally seems to be more accurate it would make sense to make it the default rather than `ellipsoid `as it is today ... or save the setting which it doesn't do currently.

b14ckyy, 2023-11-16:
Actually I did not find a location where ellipsoid had any acceptable deviation. The best I found was 20m off from other providers. With sea level I was always within a 5m or less range from what OSM and Google provided. So yes I highly recommend that this will be default. If someone really wants to switch that for whatever reason of it there are indeed regions, keep the switch with a save setting but sea level should be default for sure imho.

closed 2023-11-27 by breadoven

cptX, 2023-12-03:
In my tests in several locations in Greece I found the ellipsoid model more accurate than the sealevel (DEM option enabled in inav). So for me at the moment this option will remain disabled. I have already found significant differences between Google Earth and Bing maps heights and I'm investigating this, but with this options things get worse here...

breadoven, 2023-12-03:
Probably easiest to just to save the setting. But Sea Level is likely the better default given it tends to be more accurate closer to sea level where maybe most people fly. Ellipsoid tends to be better as you go higher.