iNavFlight/inav-configurator #1895: Configurator Modes tab and Battery

miralex773, 2023-11-18:
Using INAV configurator 6.1.0. On Modes tab, NAV modes don't show up if GPS is not detected. On my setups usually GPS and MAG are powered from 5V (not 4V5) and thus GPS is not available when setting up the aircraft with USB connection, without battery. So I have to connect the battery every time I want to make changes in Modes. Could you please make NAV modes available in Configurator when GPS/MAG are configured, but not powered/detected.

DXTDXT2020, 2024-02-19:
maybe.[pdf](https://www.baidu.com/)

DXTDXT2020, 2024-02-19:
maybe:[pdf](https://www.baidu.com/)

DXTDXT2020, 2024-02-19:
maybe:[pdf](https://d0.ananas.chaoxing.com/download/fe09b46ceb47ce3d84103fef4653fca5?at_=1708329980621&ak_=9a8b46a3ce16846bbffe0c733920dd2b&ad_=6711a6978c8c5cc67845d8845c3904d6&fn=2024+E-Organic+Chemistry-Keys)

closed 2024-05-09 by DzikuVx