iNavFlight/inav-configurator #1896: RC2 gives multiple issues

0crap, 2023-11-20:
- Battery profiles do not update when switching
- Settings validated checkmark stays red (even when known good diff all is applied)
- After full erase and flash it hangs after apply defaults for wing

RoadyFPV, 2023-11-22:
same here it shows reboot ez tune tab is there. after u close configurator and connect again ez tab is not there. and settings validated is red. cli status show ez_filter_hz error

RoadyFPV, 2023-12-06:
same in release version

0crap, 2023-12-06:
> same in release version
> 
> stuck in: Device - Saving default settings

Confirmed here as well.
That is just sloppy.
No reason not to do a RC3 for the Configurator to see if this was fixed.
Now it's still there in the final release. Nice...

Yury-MonZon, 2023-12-06:
Same here on linux x64 7.0 version Matek F411WTE

RoadyFPV, 2023-12-07:
its not a cosmetic error, after u click on x and start configurator again ez tab is missing

RoadyFPV, 2023-12-07:
thanks for the clarification. in the rc version, ez tab was also displayed for airplane. i will check it again.
and the boot stuck on savig defaults?

0crap, 2023-12-08:
>.... and the boot stuck on saving defaults?

This is still a bug as mentioned in the opening post.

closed 2024-05-06 by DzikuVx