iNavFlight/inav-configurator #1897: "Use Sea Level Earth DEM Model" for BING Maps: Broken for Wayppoints

Scavanger, 2023-11-20:
The option is not actually usable. If this option is activated and a new waypoint is created, although "Sea level ref" is deactivated, the standard elevation is added to the ground elevation; if "Sea level ref" is activated, the ground elevation is added AGAIN. In addition, it is no longer possible to specify a relative altitude, even if "sea level ref" is deselected. 
Error message: "Altitude below ground level. Change ignored".

The whole "DEM model" for Bing has to be revised.
See also: #1893

breadoven, 2023-11-21:
You need to think again about what those settings are doing and what affect they have on the altitude that's required for the WP especially depending on whether it's using absolute or relative altitude (Sea Level Ref. setting). I assume you're setting the Home position near to where the WPs are being placed rather than leaving it on Null Island (Lat/Lon = 0, 0) ?

The only issue with the Bing DEM model is the variation in elevations it uses depending on whether the Ellipsoid or Sea Level option is used. It only affects the WP altitude calculated by Mission Control when the Absolute setting is used. It has no affect when the Relative setting is used other than to change the elevations for Home and the WP but the Altitude/ground clearance is  unaffected because it's using a relative reference that cancels out the elevation changes.

The only issue that didn't get resolved last time I did this is the fact that the WP Altitudes don't auto update if you move the Home position on the map. You need to manually update them simply by selecting each WP. Although I seem to remember thinking this is perhaps a better way of doing things because if you move the Home position it changes the ground clearances meaning you need to recheck the entire mission again to make sure it's still OK in terms of ground clearance. This should be mentioned somewhere but unfortunately there is no manual for Configurator as such.

breadoven, 2023-11-22:
Pretty sure the reason elevation was added to the WP altitude when switching to Absolute reference was simply because it avoids altitudes below ground level when switching back to the Relative reference. However, this isn't an issue now since altitude/elevation sanity checking was added which automatically corrects for altitude settings that are below ground level. So it makes sense to just set WP altitude to ground elevation + default altitude setting and let the altitude/elevation sanity check correct the altitude for ground clearance when switching back to the Relative setting.

closed 2025-01-18 by b14ckyy