iNavFlight/inav-configurator #1914: all three gps sattelites get disabled when enabling BeiDou

tippfehlr, 2023-12-07:
When I try to enable BeiDou Sattelites (CN) via GPS tab or cli, after rebooting all three, Galileo, BeiDou and Glonass are disabled again (in ui and in cli).

**Steps to reproduce:**
1. enable Galileo and Glonass and reboot: works as expected
2. enable BeiDou Sattelites and reboot: all three are disabled again

**expected result:**
all three stay activated

---

```
CONFIGURATOR: 7.0.0
FC FIRMWARE: 7.0.0 [SPEEDYBEEF405WING]
```

tippfehlr, 2023-12-07:
probably an issue with INAV itself.

MrD-RC, 2023-12-07:
I’m sure this is expected behaviour. Either your GPS module doesn’t support having 4 GNSS. Or I believe I’ve read that BeiDou and GLONASS share something, that means they can’t be selected at the same time. 

Maybe Configurator needs some kind of warning?

@mmosca could we get some clarification?

mmosca, 2023-12-08:
> Alright, reopened this one and closed the issue in inav.
> If indeed glonass and BeiDou share something, maybe disable the other switch to indicate that only one is supported.

The valid combinations are defined by the gps and are not known to inav. If we add that to the configurator it may prevent other valid use cases.

MrD-RC, 2023-12-08:
I was meaning just an info box for explaining that. Not actually denying enabling certain clusters. Is that documented anywhere?

mmosca, 2023-12-08:
The data sheet for each gps model. The status command does report the maximum number of constellations a gps supports though, and I am sure that his gps does not say 4.

tippfehlr, 2023-12-08:
```
# status
INAV/SPEEDYBEEF405WING 7.0.0 Dec  5 2023 / 11:06:36 (895a4f31)
GCC-10.3.1 20210824 (release)
System Uptime: 44 seconds
Current Time: 2041-06-28T03:04:00.000+02:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
ESC Temperature(s): Motor Count = 2
ESC 0: 0°C, ESC 1: 0°C, 
SD card: Manufacturer 0x1b, 994816kB, 04/2008, v1.0, '00000'
Filesystem: Ready
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 9541, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 4, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 3, used = none
System load: 12, cycle time: 1035, PID rate: 966, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: MAX7456 [30 x 16]
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 0/1
    Glonass 1/0
    Max concurrent: 3
```

> Max concurrent: 3

ok, so my gps (HGLRC M100 Mini) does indeed only do 3 at a time.
And BeiDou and Glonass don’t work at the same time, doesn’t matter is Galileo is enabled or not.

- If this is confirmed, wouldn’t disabling the other switch make it clear?
- And I don’t know if there are gps with less than 3 max concurrent gps systems, if yes maybe display used/max gps systems.
- I’m in Europe/Germany, does it matter which one I activate? Or should I disable both?

mmosca, 2023-12-08:
That is a hard thing to say these days. Glonass is provided by a country
that is engaged in a war in europe. beidou has global coverage, but works
better in the asia pacific region.

I would say that your gps defaults of galileo and beidou is probably a good
combo.

mmosca, 2023-12-08:
> I have same problem. I have Config 7.0.0 and Inav 7.0.0. Can only switch on 2 of 3: ![grafik](https://private-user-images.githubusercontent.com/18383738/289118785-74b57b0a-6a6d-4530-bc27-68c491717ca6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwNTUwMTgsIm5iZiI6MTcwMjA1NDcxOCwicGF0aCI6Ii8xODM4MzczOC8yODkxMTg3ODUtNzRiNTdiMGEtNmE2ZC00NTMwLWJjMjctNjhjNDkxNzE3Y2E2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA4VDE2NTgzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjZTc2MjE1NWJhY2YzMjQwNjE1YTM3N2YxNThlOTY3ODk5NzMwMmFlMzc4YTViODc0OWQ4NjI5OGY0NTQ3ZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PnUB7cSXbLmGJ9Jv6jiPtzgXmdwS-gXB5kG8kl8KJoA)

Normal, and expected. Just read the rest of the discussion.

tippfehlr, 2023-12-08:
I think showing the actual capabilities of the gps unit is much more user friendly than letting the user figure out what works and what doesn’t.
Is there a reason you are against disabling the options? The note can still be in the tooltips.

mmosca, 2023-12-08:
There is no way to find out before hand what combinations are valid, but if
you do find a way in the ublox documentation, please share and we may look
into improving the configurator behavior.

tippfehlr, 2023-12-08:
That is true, I was assuming that it was a limitation of the gnss itself (without having done research -_-)
A note is fine I guess

b14ckyy, 2025-01-18:
Guess this one was forgotten :)

mmosca, 2025-01-19:
I think @sensei-hacker has addressed this. 

@sensei-hacker should we close this?

sensei-hacker, 2025-01-19:
Closing because this was addressed in #9646 for INAV 7.1.0

The M10 smartwatch chip has only one single receiver. As Mr D noted, Beidou and Glonass use the same channel.  With only one receiver, the M10 low-power chip can only listen to one constellation on that channel.

The workaround is to use the low-precision channel for M10. All others - M8, M9 use the high precision channel because they have multiple receivers.

closed 2025-01-19 by sensei-hacker