iNavFlight/inav-configurator #1924: Flywoo 405 pro motor issue

captrambo, 2023-12-12:
Flashed Flywoo F405 Pro on F405 versatile board as it is only target available.
ESC is flashed with blueJay same was working on Inav -6 but on inav7 esc are not detected same is not detected on ESC configurator.
Flashed ESC to BL Heli S still same result.
On RC2 behaviour was same but with two motor
Wiring is not issue same flashed with Betaflight everything in order

frank-hubgit, 2023-12-17:
same problem here with the flywoo explorer using this stack.
Going back to betaflight all 4 motors are working.
Using inav 6.1 everything was fine.
By the way: Where can I get the INAV-Configurator 6.1.1 linux_64 again in order to check it all 4 motors are working with this previous version? I will never delete an old installation-file again...
Thanks in advance.

frank-hubgit, 2023-12-17:
I have got the inav-configurator version 6.1.0.
After going back to this version all 4 motors are working well.
the esc-configurator.com is detecting all 4 esc.

mmosca, 2024-02-23:
Check the mixer tab and make sure the timer for the first 4 outputs is set to motor.

frank-hubgit, 2024-02-23:
Thanks for your tip. That works great.
Kind regards
Frank

closed 2024-02-23 by mmosca