iNavFlight/inav-configurator #1926: Demo mode not working on INAV Configurator 7.0.0 on macOS 12.7

AndreaPro, 2023-12-18:
I've installed from the .dmg and from .zip, this is the first time that I install INAV. When selecting Demo mode and clicking Connect in the logs I get:
`Failed to open MSP connection`
On the Telegram group I've asked for help and someone told me to press "Reset Demo Mode" in the preferences. I've tried but this appears in the logs:
`Unable to reset Demo mode: ENOENT: no such file or directory, stat '/Users/Andrea/Library/Application Support/inav-configurator/Default/demo.bin'`

Inside System Preferences > Security and Privacy I have provided Full Disk Access but nothing changed. Any idea?

stronnag, 2023-12-19:
Demo Mode requires the `inav_SITL`, which is only available on Linux and Windows. 
This is expected, albeit non-intuitive, behaviour.

AndreaPro, 2023-12-19:
Ok, then it could be a good idea, if possible, to inserted a line in the logs reporting that the feature is not supported on macOS

closed 2024-04-24 by DzikuVx