iNavFlight/inav-configurator #1927: Inav 7 is not working with speedy bee f405 v3

MusafirFpv, 2023-12-25:
Hello this is Vijay 
and I’m trying to install inav7 on Speedybee f405v3 stack  but despite assigning port  and Esc protocol for Esc,  esc not getting calibrated and after assigning port for smart audio all assigned ports get disabled.
So I tried inav 6 version esc working properly but same issue occurring while assigning smart audio port that is disabling all assigned ports. Any one have solutions for this

MrD-RC, 2023-12-25:
If the portd are getting forgotten, it is due to an illegal port assignment. Please can you post a screen grab of how you are trying to set up your ports.

MusafirFpv, 2023-12-25:
@MrD-RC 
Ya sure but with same ports sequence assigned on betaflight working perfectly even on INAV 6 except smart audio port .

MusafirFpv, 2023-12-25:
Well which is the best FC compatible with INAV

MrD-RC, 2023-12-25:
It’s probably what you’re selecting on the ports page. Which is why I’ve asked for the screen grab. BF isn’t the same for all assignments. So don’t use that as a comparison.

MusafirFpv, 2023-12-27:
@MrD-RC ** @Thanks for your support** after assigning on different Uart for smart audio working fines.Esc requires calibration and all issues solved 
Just figuring out how control vtx power with radio

closed 2023-12-27 by MusafirFpv