iNavFlight/inav-configurator #1937: Issue: cannot connect after compiling and flashing INAV from Target  STM32F405 and STM32F446 boards

synersignart, 2024-01-02:
INAV7.0.0
Chipset : STM32F405RGT6  and  STM32F446RET6
Oscillator Crystal: 8mhz

Issue: cannot connect after compiling and flashing INAV from Target

STM32F405 missing in Target folder  _if defined_  **system_stm32f4xx.c**
STM32F446 Discrepancy  in  Target folder  system_stm32f4xx.c and Link Folder **stm32_flash_f446xe.ld** stating defined (STM32F446xx) VS stm32_flash_f446xe.ld file

would like to confirm is any one has issues in compiling , flashing and connecting 

[image]
[image]

mmosca, 2024-01-03:
I suspect it may be a different issues. If the problem was missing definition for the MCU, it would not biuld and fail with the "Undefined CPU" error.

If the issue is connecting, it could be something else. What does your CMakeFiles.txt looks like? I believe the default value for the HSE cristal is 8Mhz. You can try adding HSE_MHZ 8 to your target, just to double check.

synersignart, 2024-01-04:
I have to check , if there are some conflicts