iNavFlight/inav-configurator #1939: Rovers & boats that are nav-capable are presented(what seems to be) mostly irrelevant params in "Advanced Tuning"

packetpilot, 2024-01-12:
...and to make the learning curve a whole lot steeper, those that _are_ relevant lack any indication that they're in the rover loop (as they're all labeled fixed wing).

Are rovers & boats destined to remain third-class citizens for the foreseeable future?

Jetrell, 2024-01-13:
> Are rovers & boats destined to remain third-class citizens for the foreseeable future?

Yes, this will likely be the case for a while.

As you've experienced, You have to wade through the settings to see which ones can be applied to your usage case.
Altitude navigation features are obviously not used.. But having altitude on the OSD, to tell you how far you have ascended or descend above your home location while covering terrain, is handy to know for the RF signal quality, of both Video and RC link.

sensei-hacker, 2024-01-13:
So far,  each boat, rover, backhoe, etc has been different.
There just isn't any such think as a "standard model submarine" or "standard rover".

Taking away options from you because I don't use them would be doing you a disservice.

As examples, someone who makes and airboat may think that pitch and altitude don't apply. The next person who comes along is making ski boat and pitch is VERY important on a ski boat or speed boat. (As I learned in my full sized 28 foot boat).

The next person who comes along very much wants to control the altitude of the boat, which normally cruises about 50cm below the surface.

So yeah we probably never will get rid of the features we don't expect a typical rover or a typical watercraft will use, because your rover probably isn't typical.

packetpilot, 2024-01-13:
This very helpful explanation is indeed enlightening, and I suppose even
the multirotor parameters might apply to a submersible or maybe even zany
"airrover" that has no motor driving the wheels but instead props on quad
arms.

Would it be conceivable and if so, reasonable to filter displayed Advanced
Tuning params based on conditional logic that considers the current mixer?
e.g. if the craft has only a single motor output, then MR advanced tuning
params would be hidden (or at least greyed)?

If not, then perhaps I should ask (maybe in a discussion and not this
issue) which subset of parameters in advanced tuning apply to what I
consider a "traditional" rover: a single direct drive propulsion channel
and a single direct drive yaw (or steering) channel. It seems some of the
fixed wing parameters apply. Unsure if it's all of them, and there seems to
be no source for this info other than the literal source code, despite a
big YT rover content surge a few years back that afaict doesn't involve
navigation/waypoint missions.

I thank you for helping me escape my own context -- we user plebs down here
certainly don't have the use case awareness that the devs do, and I'm very
appreciate of your help individually from your response, and if course
collectively for the sweet goodness INAV delivers.

closed 2025-01-18 by b14ckyy