iNavFlight/inav-configurator #1940: GCC 13.2 and profile/mixer, increasing settings load lockup occurrence

Jetrell, 2024-01-12:
The matter of DIFF/DUMP introduced settings errors, or locking up the compiler when loading, is not new, when the DIFF ALL has multiple `PID` and `Battery` Profiles applied.. This has been evident with F411's, since back in the 3.0 releases.

But this is also now experienced in 7.0, with F405's, since the addition of the Profile mixer switch.
And now that the compiler has been upgraded for 8.0. Its even starting to effect F722 controllers when loading settings.

In my experience it does not occur every time. So it may not be seen by all users.. But for someone as myself who is often flashing fw and loading settings for testing purposes. Its has become far more noticeable. With the only workaround, being to load the DIFF ALL in two parts.

It would appear there is a fixed transfer rate between the Configurator and the flight control ? Due to it effecting processors in order of their clock speeds, with the introduction of more demanding background tasks being placed upon them.

stronnag, 2024-01-12:
To add a bit more context, following some discussion with @Jetrell 

* The hack in the configurator to use arbitrary delays is an inadequate solution, masking the real issue.
* The problem is serial overrun due to the configurator not "pacing" the delivery of data to that the FC can process.
* For example, `profile X` takes c. 1 second to return to a CLI prompt. The configurator can swamp the FC during this time.
* mwp's `fcset` tool does not use a delay, instead it awaits the FC sending the prompt "\n# " before sending the next line, so it can't overrun the FC's serial processing.
* @jetrell has reported that a further configurator hacked to increase the delays does alleviate the immediate issues. 
* While increasing the extant delays from 50ms to 100ms for normal line delay and from 200ms to 1000ms for `profile` delay did alleviate problem, it would, in this reporter's view, be preferable to adopt an "await prompt" solution. 
* The "await prompt" solution has the additional benefit of increasing throughput as it does not incur delay when it is not necessary.

rts18, 2024-01-14:
I concur with your observations gentlemen. I too have experienced the matters. Both loading 2 or 3 profiles. And more so of late, since the GCC13.2 tool chain commit.

stronnag, 2024-01-22:
Just for "fun", I instrumented mwptool's `fcset` to measure the time elapsed from writing a line command to receiving the new CLI prompt "\n# ".  This is on a F405 bench test FC,  recent INAV master (8.0), demonstrating that the current delays are an inadequate provision against overrun.

| Index | Time(s) | Command |
| ---- | ---- | ---- |
| 0001 |  1.207 | defaults |
| 0002 |  0.048 | servo |
| 0003 |  0.016 | servo |
| 0004 |  0.016 | feature |
| 0005 |  0.016 | feature |
| 0006 |  0.016 | feature |
| 0007 |  0.016 | feature |
| 0008 |  0.016 | feature |
| 0009 |  0.016 | beeper |
| 0010 |  0.016 | blackbox |
| 0011 |  0.016 | blackbox |
| 0012 |  0.016 | blackbox |
| 0013 |  0.016 | blackbox |
| 0014 |  0.016 | blackbox |
| 0015 |  0.016 | blackbox |
| 0016 |  0.016 | blackbox |
| 0017 |  0.016 | blackbox |
| 0018 |  0.016 | blackbox |
| 0019 |  0.016 | blackbox |
| 0020 |  0.016 | blackbox |
| 0021 |  0.016 | blackbox |
| 0022 |  0.016 | blackbox |
| 0023 |  0.016 | serial |
| 0024 |  0.016 | serial |
| 0025 |  0.016 | serial |
| 0026 |  0.016 | serial |
| 0027 |  0.016 | serial |
| 0028 |  0.016 | serial |
| 0029 |  0.016 | serial |
| 0030 |  0.016 | aux |
| 0031 |  0.016 | aux |
| 0032 |  0.016 | aux |
| 0033 |  0.016 | aux |
| 0034 |  0.016 | aux |
| 0035 |  0.016 | aux |
| 0036 |  0.016 | aux |
| 0037 |  0.016 | aux |
| 0038 |  0.016 | aux |
| 0039 |  0.016 | aux |
| 0040 |  0.016 | aux |
| 0041 |  0.016 | set gyro_zero_x |
| 0042 |  0.016 | set gyro_zero_y |
| 0043 |  0.032 | set gyro_zero_z |
| 0044 |  0.016 | set ins_gravity_cmss |
| 0045 |  0.016 | set acc_hardware |
| 0046 |  0.016 | set acczero_x |
| 0047 |  0.016 | set acczero_y |
| 0048 |  0.016 | set acczero_z |
| 0049 |  0.032 | set accgain_x |
| 0050 |  0.032 | set accgain_y |
| 0051 |  0.016 | set accgain_z |
| 0052 |  0.032 | set mag_hardware |
| 0053 |  0.032 | set baro_hardware |
| 0054 |  0.032 | set serialrx_provider |
| 0055 |  0.032 | set rx_min_usec |
| 0056 |  0.032 | set blackbox_rate_denom |
| 0057 |  0.032 | set failsafe_procedure |
| 0058 |  0.032 | set align_board_roll |
| 0059 |  0.032 | set align_board_pitch |
| 0060 |  0.064 | set small_angle |
| 0061 |  0.064 | set applied_defaults |
| 0062 |  0.064 | set gps_provider |
| 0063 |  0.064 | set gps_sbas_mode |
| 0064 |  0.064 | set gps_ublox_use_glonass |
| 0065 |  0.096 | set nav_wp_radius |
| 0066 |  0.096 | set nav_wp_max_safe_distance |
| 0067 |  0.112 | set nav_rth_allow_landing |
| 0068 |  0.112 | set nav_rth_altitude |
| 0069 |  0.112 | set nav_rth_home_altitude |
| 0070 |  0.112 | set nav_fw_bank_angle |
| 0071 |  0.112 | set nav_fw_loiter_radius |
| 0072 |  0.112 | set fw_loiter_direction |
| 0073 |  0.112 | set nav_fw_launch_detect_time |
| 0074 |  0.112 | set nav_fw_launch_motor_delay |
| 0075 |  0.112 | set nav_fw_launch_max_altitude |
| 0076 |  0.112 | set nav_fw_launch_climb_angle |
| 0077 |  0.112 | set nav_fw_allow_manual_thr_increase |
| 0078 |  0.112 | set telemetry_halfduplex |
| 0079 |  0.112 | set smartport_fuel_unit |
| 0080 |  0.160 | set name |
| 0081 |  0.016 | mixer_profile 1 |
| 0082 |  0.048 | set model_preview_type |
| 0083 |  0.032 | mmix |
| 0084 |  0.016 | mmix |
| 0085 |  0.016 | mmix |
| 0086 |  0.016 | smix |
| 0087 |  0.016 | smix |
| 0088 |  0.016 | smix |
| 0089 |  0.016 | smix |
| 0090 |  0.016 | smix |
| 0091 |  1.266 | mixer_profile 2 |
| 0092 |  0.016 | profile 1 |
| 0093 |  0.080 | set fw_p_pitch |
| 0094 |  0.080 | set fw_i_pitch |
| 0095 |  0.080 | set fw_ff_pitch |
| 0096 |  0.064 | set fw_p_roll |
| 0097 |  0.080 | set fw_i_roll |
| 0098 |  0.080 | set fw_ff_roll |
| 0099 |  0.080 | set fw_p_yaw |
| 0100 |  0.064 | set fw_i_yaw |
| 0101 |  0.080 | set fw_ff_yaw |
| 0102 |  0.080 | set max_angle_inclination_rll |
| 0103 |  0.080 | set max_angle_inclination_pit |
| 0104 |  0.048 | set tpa_rate |
| 0105 |  0.064 | set tpa_breakpoint |
| 0106 |  0.048 | set roll_rate |
| 0107 |  0.064 | set pitch_rate |
| 0108 |  0.048 | set manual_rc_expo |
| 0109 |  1.264 | profile 2 |
| 0110 |  1.260 | profile 3 |
| 0111 |  0.016 | battery_profile 1 |
| 0112 |  0.032 | set bat_cells |
| 0113 |  0.032 | set vbat_min_cell_voltage |
| 0114 |  0.048 | set vbat_warning_cell_voltage |
| 0115 |  0.048 | set battery_capacity |
| 0116 |  0.048 | set battery_capacity_warning |
| 0117 |  0.048 | set battery_capacity_critical |
| 0118 |  0.048 | set throttle_idle |
| 0119 |  0.048 | set nav_fw_cruise_thr |
| 0120 |  0.048 | set nav_fw_max_thr |
| 0121 |  0.064 | set nav_fw_launch_thr |
| 0122 |  1.257 | battery_profile 2 |
| 0123 |  0.032 | set bat_cells |
| 0124 |  0.048 | set vbat_min_cell_voltage |
| 0125 |  0.048 | set vbat_warning_cell_voltage |
| 0126 |  0.048 | set battery_capacity |
| 0127 |  0.048 | set battery_capacity_warning |
| 0128 |  0.032 | set battery_capacity_critical |
| 0129 |  1.262 | battery_profile 3 |
| 0130 |  1.266 | mixer_profile 1 |
| 0131 |  1.264 | profile 1 |
| 0132 |  1.267 | battery_profile 2 |
| 0133 |  6.067 | save |

rts18, 2024-06-27:
@DzikuVx Why was this closed? The issue @stronnag has described in the previous posts has not been resolved. 

>The problem is serial overrun due to the configurator not "pacing" the delivery of data to that the FC can process.
>For example, profile X takes c. 1 second to return to a CLI prompt. The configurator can swamp the FC during this time.

Using the latest master from both repositories and loading settings with more than one profile, the configurator still locks up.
Requiring the 'diff all' to be painfully loaded in multiple parts.

stronnag, 2024-06-28:
Reopening as multiple users report this issue, which does not appear to have been addressed.

silicongarage, 2024-11-18:
I've hit this problem with version 7.1.2  on  Speedybee F405 Wing.