iNavFlight/inav-configurator #1945: Not recieving sonar values from 3901-L0X

EclipseMantis, 2024-01-23:
Hello all, 
I'm trying to use a 3901-L0X opflow and lidar sensor to enable position holding in environments without GPS. I have it connected and can read opflow data from it. However, the sonar value is static at -1cm regardless of what happens around it. It is lit up blue in the dashboard, so it should be working (?). 

[image]

I flashed ArduPilot to the board and the sensor worked great, so it is a software issue or user error. Regardless, I'd really prefer to use iNav.

Any thoughts?

EclipseMantis, 2024-02-08:
I am still working on this and would appreciate help 👍

jureopacak, 2024-03-07:
Just check your FC alignment. If your FC is tilted more than some angle INAV will disable the sensor per default.

Place your FC on the flat surface and try again.