iNavFlight/inav-configurator #1949: Need several attempts to connect FC to INAV 7.0.1 on Macbook Air M1

jonathanoeijoeng, 2024-01-27:
Really often I have to plug and unplug usb cable several time to connect FC to INAV on Macbook Air M1
Even most of the time I have to restart macbook and then it can connect to INAV.
Using:
- Macbook Air M1
- INAV 7.0.1 
- FC Speedybee F7 v3
- USB cable usb A to USB with dongle / usb c to usb c

Any solution or advice?

Jonathan

mszulen, 2024-05-19:
Same here. Power cycling not always work. It's trully a lottery now.
A few times I've noticed that powering on the FC and then leaving it for longer time (a few minutes even) does allow me to connect without error.

closed 2025-01-18 by b14ckyy

siliconx, 2025-04-13:
The same issue (inva8, speedybee f405 mini, macbook air m4), only plug in the battery can connect successfully.