iNavFlight/inav-configurator #1972: 7.1 RC1 - SITL build outdated and locks up when applying presets

b14ckyy, 2024-02-19:
The SITL Build in the 7.1 Configurator is based on a 7.0 Pre-Release (or RC) build Nov 19 2023 / 17:40:01 (dc19ce5a). 
When applying a preset it locks up the configurator in a waiting loop. 

Applying a fixed wing preset to a 7.0 Release build Dec  5 2023 / 09:44:21 (895a4f31) works fine though. 

SITL should be updated to the firmware release build as well.

Scavanger, 2024-02-19:
@DzikuVx 
Would it be possible to include the SITL executables in the assets on GitHub with the INAV releases?
Then we could easily integrate an autoupdater in the SITL tab.

b14ckyy, 2024-03-27:
7.1 final has a relatively current build of 7.1 SITL but already 17 days old.

mmosca, 2024-04-08:
Yes, they are generated in normal github builds

mmosca, 2024-04-08:
> @DzikuVx Would it be possible to include the SITL executables in the assets on GitHub with the INAV releases? Then we could easily integrate an autoupdater in the SITL tab.

They are build as part of normal github builds, so they can easily be added to the releases as well

closed 2025-01-18 by b14ckyy