iNavFlight/inav-configurator #1975: eByte M10 GPS not working with INAV

DM4DS, 2024-02-21:
Hi,

tryed it with 7.0 and now with 7.1 RC1 and the GPS modules are not working.
They work perfectly with the UBLOX software on a FTDI adapter, but not with INAV.

The GPS symbol in the software stays RED. Tryed multiple baudrates and also ublox7 mode, still red symbol.

Is this known? Was not able to find something.

They are "cheap" compared to other devices and work great for me in other aplications. Would be great to use them also in INAV.

If you need infos, let me know. I also own measure equipment, if this helps to find a solution.

The exact modell: eByte E108GN02D

Thanks for your help
DM4DS - Sandor

sensei-hacker, 2024-02-21:
M10 modules generally work fine.
In 7.1 they can do four constellations; in 7.0 three constellations.

Please show your GPS config, your ports config, and your wiring.

DM4DS, 2024-02-21:
Hi,

wires is fine, a photo of 4 wires will show the same.
An older Ublox 7 and a M10 from HGLRC works fine, but not the eByte M10. Which works on a FTDI perfect, so the module is not dead. Also tried a second eByte M10, the same.

GPS config is as always, tryed as I wrote different baudrates, too .

In the GPS menu, as you aksed, normally the statistics are showing increasing numbers, but not on the eByte. that´s the only difference I can see, so there is no communication with the module.

MrD-RC, 2024-02-21:
It is an NMEA GNSS module. It doesn't support UBLOX. INAV doesn't support NMEA.

closed 2024-02-21 by MrD-RC

DM4DS, 2024-02-21:
INAV does not support NMEA at all? ok.

The module works fine with all my other things.

Thanks @MrD-RC

MrD-RC, 2024-02-21:
> INAV does not support NMEA at all? ok.
> 
> The module works fine with all my other things.
> 
> Thanks @MrD-RC

No. It was deprecated in version 6.0 and removed in 7.0.

DM4DS, 2024-02-21:
@sensei-hacker my units are M10 according to the ublox software and work file. Maybe they updated the module but not the PDF.

@MrD-RC ah, that's the reason why they work on the older stuff. Thanks

MrD-RC, 2024-02-21:
They're not M10 GNSS modules. M10 is UBLOX. These modules aren't UBLOX. They are NMEA0183 V4.1.