iNavFlight/inav-configurator #1985: No way to correct FC orientation as it is on the airframe - fixed wing

Mafracao, 2024-03-06:
I'm currently building an old Tsunami F3J (high end RC glider) into a FPV platform. I'm using Matek F405 WMN FC and iNav 7.

Due to physical space issues, the FC cannot be placed with the arrow pointing to the front of the plane and, as you can imagine, as it is right now I can't find a way to properly set the FC orientation on the configurator. If I set the FC orientation into the configurator as it is on the real plane, the simulation shows the plane upside down. 

Is this a bug or the FC MUST have its arrow pointed to the front no matter what? Is there a work around it? I'm very used to building multi rotors, and this has never been an issue before (although I always do my best to build the FC in the "default" orientation).

Thanks and best regards;
Mafracao

sensei-hacker, 2024-03-06:
> If I set the FC orientation into the configurator as it is on the real plane, the simulation shows the plane upside down.

Flip the roll orientation you have upside down from how you have it now. Then if the left and right are reversed, rotate yaw 180°.

Mafracao, 2024-03-07:
> > If I set the FC orientation into the configurator as it is on the real plane, the simulation shows the plane upside down.
> 
> Flip the roll orientation upside down from how you have it now. Then if the left and right are reversed, rotate yaw 180°.
> 
> Any orientation can be set. It can be tricky to think about because reversing one axis effects the other. So, like a Rubik's cube, in order to effect only one axis, you have change two - the axis you want to change, then reverse the implied change on the other axis.

I wish it was that easy, but trust me: no matter what is done into the configurator, there is no way the SETUP page and the ORIENTATION page will be correct at the same time (if the FC has it's arrow pointed backwards, as it is on my plane).

I've spent a good couple of hours re-doing calibrations and trying all possible positions into the ORIENTATION page. The best results I've got showed into the SETUP page either as the plane is upside down or with reversed pitch.

I've actually set it in such way that the plane animation was 100% correct on the SETUP page, but to do so the ORIENTATION page was not representing what was in the plane. It flew ok, but as soon as I switched to test the Return To Home the plane flipped violently upside down (my thinking is that the FC though I was flying backwards).

sensei-hacker, 2024-03-07:
The sample flight controller on the orientation page does not matter. What matters is how the model moves on the Setup page.

If you'd like to jump on the Discord I can walk you through it. I'm working on an easier interface for a future version of INAV Configurator.

Mafracao, 2024-03-07:
Thank you for the help. Well, with that in mind, I'm able to make the animation on the SETUP page look correct.

The problem is: I've done it before, and as soon as the RETURN TO HOME was switched on, the plane literally tumbled on the sky (it gave full elevator down, making a half inverted loop). After that I switched it back to manual mode and flew it home by myself.

Can this be related to a wrong compass setup? Something else?

Apart from that, the other odd behavior I've noticed was the barometer being way off, giving measures into the OSD that would be too low or to high, with odd variations as well...

I'm sorry for presenting so much trouble. I have a fixed wing INAV plane that works flawlessly, and this is my second build. On this second one, I'm having nothing but trouble. It's on its second FC, since the first one simply would go crazy and stop responding whenever I plugged flaps on the S8-S10 ports....

I'm kind of a noob on this, so how must I find you on discord? I don't really use it to be honest...

Again, thank you so far.

sensei-hacker, 2024-03-07:
Compass issues will not cause the behavior you described. Sounds like it's most likely upside down and backwards, which will result in similar behavior.

Our Discord is here:
https://discord.gg/BfRnpkH8

sensei-hacker, 2024-03-07:
Btw, when comparing with the onscreen model, have the aircraft pointed toward the screen. The model should show red on the left, green on the right.  If green is on the left, it's upside down.

MrD-RC, 2024-03-07:
You can set the board orientation in both the Configurator UI and in CLI. This appears to be a setup issue. So I’m converting it to a discussion n