iNavFlight/inav-configurator #2010: Can not find NAV Modes in inav-configurator 7.1.0

wang322588, 2024-04-01:
I have set the GPS and baro (without Mag which allowed in version 7.1.0) but can not find the mode memu.
[image]

Sil20, 2024-04-04:
One additionnal remark : FAKE mag must be disabled after MODE setup, because it brings out a hardware alerte when powering the FC

wang322588, 2024-04-04:
If a pilot convert from betaflight to Inav with a none Mag GPS preinstalled.  This setting process may be disturbing like you said.

MrD-RC, 2024-04-04:
This is actually in the release notes. Of folks read them, they know that it has to be set up with a mag at the moment. This has already been fixed by https://github.com/iNavFlight/inav/pull/9824 and will be in a future release. 

I think too many pilots come over from BetaFlight and assume they know how everything works. INAV has a similar Configurator. That is all. They should learn how to use INAV properly. Not just guess. The number of people who have problems setting up MSP DisplayPort, because they try to use BetaFlight’s settings is crazy.

closed 2024-04-04 by MrD-RC

Sil20, 2024-04-04:
Well, 
I've been using INAV for many years on fixed wings and the configurator looked suspicious. it has been fixed so it means it is considered as an unconvenience at least. 
Using FAKE compass then revert to no compass and losing NAV mode in the MODE panel, is not described in the INAV release note :(https://github.com/iNavFlight/inav/wiki/7.1.0-Release-Notes)

Good thing it is sorted good work!

MrD-RC, 2024-04-04:
> Well, 
> 
> I've been using INAV for many years on fixed wings and the configurator looked suspicious. it has been fixed so it means it is considered as an unconvenience at least. 

This does not affect fixed wing at all. They have never needed compass for navigation. It only affects multirotors. They DID require compass for navigation. 

How does Configurator look suspicious!?

Sil20, 2024-04-04:
Once the RTH and NAV modes are activated thanks to the Fake mag, removing the fake Mag remove also the RTH and NAV modes from the MODE panel. 
Though those RTH and NAV modes are still functionnal unless I am mistaken.

Sil20, 2024-04-04:
Canot find the NAV mode --> look at the documentation/comments find the FAKE mag workaround

Enable FAKE mag
Enable NAV mode --> FC hardware alarm related to MAG ==> suspicious ==>  OK why not, but could have worked even if I could find a technichal reason had I looked in the code or documentation.

Disable FAKE MAG --> could not find the nave MODE in the nav panel ==> suspicious ==>  OK why not, but could have worked even if I could find a technichal reason had I looked in the code or documentation.

https://github.com/iNavFlight/inav/pull/9824 but This was not found in my search in the issue github database
Found this 2010 issue, add my comment to add more information, already fixed  hurray !

MrD-RC, 2024-04-04:
It is not suspicious at all. It is a way to get around a bug in the current version.