iNavFlight/inav-configurator #2012: Cannot prefetch target: No configuration received within 10 seconds, communication failed.

AcadJim, 2024-04-03:
I have downloaded iNAV and installed it on my MacBook Pro. iNAV seems to work OK up to the point of flashing my SpeedyBee F405 WING APP. After clicking on the “Firmware Flasher” tab, I am seeing messages in the “Log Box” that looks like this…

2024-04-02 @ 20:31:02 -- Running - OS: MacOS, Chrome: 98.0.4758.82, Configurator: 7.1.0
2024-04-02 @ 20:31:03 -- Loaded release information from GitHub.
2024-04-02 @ 20:33:36 -- Attempting automatic target selection
2024-04-02 @ 20:33:36 -- Connected to: /dev/cu.Bluetooth-Incoming-Port
2024-04-02 @ 20:33:46 -- Cannot prefetch target: No configuration received within 10 seconds, communication failed
2024-04-02 @ 20:33:46 -- MSP connection successfully closed

[SPEEDYBEEF405WING] is the target.
[7.1.0-SPEEDYBEEF405WING-2024-1 6:5] is version.

I click on [Load Firmware (Online)]. In the black banner shows (545426 bytes).  I click on [FLASH FIRMWARE]. Nothing happens.

I’ve working on this for several days.

Can you guys help?
Thanks

SNGFP, 2024-04-03:
2024-04-02 @ 20:33:36 -- Connected to: /dev/cu.Bluetooth-Incoming-Port
You have to connect to /dev/cu.**usb** 

I have upgraded iNav7RC (GCC-10.3.1 20210824) to final 7.1. Also configurator 7.1 on iMac and Macbook M1. I get no connection to Speedybee F7v3 since flashing to final 7.1 version (10s timeout as above). 
Back to flashing Betaflight every connection works fine - so FC should be fine. 
Also no iNav7.1 connection with Speedybee app, but still working with Bf (as before with 7.1rc). 
Flashing a Mamba FC with iNav7.1 worked fine, no connection problem...

Tell me if you need more information. 
I think the configurator has a mac/speedybee problem - but sorry i am no programmer...

AcadJim, 2024-04-04:
First of all, I want to thank you for responding to me.

I was finally able to connect my Mac (iNAV 7.1.0) to my flight controller. Turns out the connections instruction’s that came with my SpeedyBee F405 WING were wrong. Step 1 and step 2 are reversed. 

“Tell me if you need more information.” Oh yes I have a lot of questions but for now, I just have one…

How do I save a firmware configuration to my local drive?

Thanks for your help.
-Jim

> On Apr 3, 2024, at 2:50 PM, SNGFP ***@***.***> wrote:
> 
> Tell me if you need more information.

sensei-hacker, 2024-04-04:
> Step 1 and step 2 are reversed.

For the next person, what were step 1 and step 2?

To save your settings, go to the CLI tab and click ""diff all", then click "save to file".

AcadJim, 2024-04-04:
Again, thanks for your help.Actually, it goes like this for me,Open the iNAV Configurator on my computer. This gets me to the “Welcome” tab. Do not change to the “Firmware Flasher” just yet!At this point the connection download box looks like this…None of the above will works.2. With a USB-C to USB-C data cable connected to the provided USB extender, press and hold the BOOT button and connect the USB cable to the computer. The Blue “Connect” button will not work yet.Now the connection drop download box looks like this…2. Click on the “Firmware Flasher” tab. The “Auto-select Target” button does not work. Choose the “Choose a Board” drop down…[SPEEDYBEEF405WING] is the target.[7.1.0-SPEEDYBEEF405WING-2024-1 6:5] is the firmware version.I click on [Load Firmware (Online)]. In the black banner (545426 bytes).  I click on [FLASH FIRMWARE] and the flashing starts.I have to do this process every time I make adjustments with the iNAV configurator. When finished I click on “Connect” and make changes no the flight controller.As a side note, I turned on the beeper (on the USB extender). This acts as a feedback as to weather or not the flight board is finished.“To save your settings, go to the CLI tab and click ""diff all", then click "save to file".Thanks for the tip! Wouldn’t it be cool if there was a button for that! The above “save to file” saves to a *.txt file. I tried to reload that file but iNAV couldn’t read it.As I go along I’m discovering more bugs.Thanks again,-Jim

AcadJim, 2024-04-06:
Just an update, SpeedBee tech support is helping me work through these issues.They’ve have ask for…High-resolution photos of both the front and back of the wireless board and the FC (Flight Controller) board.✔️My purchase record, including the store where the item was purchased, the date of purchase, and the order ID. This information is crucial for facilitating our after-sale service.✔️A “DUMP” file from the flight controller.✔️And I’ve sent these to them.Let’s see what they do. In the meantime.Thanks for your help.

SNGFP, 2024-04-08:
Today i tried with (a virtualbox) Windows 10. Same as before, flashing final iNav7.1 worked, after that no connection possible (yes, com-ports were working). Back to Betaflight, everything is fine working with Win10. 
I still think, something is wrong with actual iNav Configurator and/or FC target...

@AcadJim : Please tell about Speedybee´s answer.

synersignart, 2024-04-16:
recently I have similar issues with my F411  having difficulty connecting after a fresh flash of firmware. can anyone explain whats going on. it was working weeks ago

TomComm27, 2024-05-07:
Hello synersignart and SNGFP,
I have also been struggling with this problem for 2-3 weeks. I also tried the 7.1.1 firmware from inav yesterday without success!
I have been liaising with Speedybee support for several days. The Speedybee F405 Wing has no errors according to the LED status says the support. I have already sent them videos and photos of what I am doing.
Have you already been able to establish a connection?
inav must have a BUG here because I can connect with betaflight and even a wrong firmware and get configurations!

Perunovskiy, 2024-06-22:
Hello guys! 

I've struggled same issue

My inav configurator 7.1.1 doesn't see my speedybee f405 wing app FC

Does anybody know solution for this issue? I see that last comment was 1,5 months ago

sensei-hacker, 2024-06-22:
Aside from the solutions listed on the firmware flasher page, also make sure you don't have the Arduino IDE open, or any 3D print slicer, or an ESC configuration tool. Those will mess with the serial port.

closed 2025-01-18 by b14ckyy