iNavFlight/inav-configurator #2033: Font manager won't open

breadoven, 2024-04-18:
Using Configurator built from the current master clicking on the Font Manager button fails to open the font window. Chrome console shows the following error:

```
script.js:90 Uncaught TypeError: r.getClientRects is not a function
    at ce.fn.init.offset (script.js:90:83947)
    at jBox.position (script.js:20770:30988)
    at jBox.<anonymous> (script.js:20771:2575)
    at jBox.open (script.js:20771:3264)
    at jBox.toggle (script.js:20771:4474)
    at jBox.<anonymous> (script.js:20770:28200)
    at HTMLAnchorElement.dispatch (script.js:90:40035)
    at HTMLAnchorElement.v.handle (script.js:90:38006)
```
Anyone else noticed this or is it related to some issue with the local build ?

breadoven, 2024-04-18:
It sounded like this could be related to an issue with `jquery `which was bumped up to version 3.7.1 in https://github.com/iNavFlight/inav-configurator/pull/1934. I reverted `jquery `back to version 2.1.4 and it now works again so it seems this is probably related to the newer version. Any ideas @DzikuVx ?

Scavanger, 2024-04-18:
Yes, the jBox version included in the Configurator is not compatible with the new jQuery version.

DzikuVx, 2024-04-22:
Looks like jbox will have to be updated then

Scavanger, 2024-04-22:
> Looks like jbox will have to be updated then

Includes in #1961