iNavFlight/inav-configurator #2050: Board Alignment not corresponding to proper control surface movement in Angle and Acro modes

jbrandwi, 2024-04-28:
This may be related to [2025](https://github.com/iNavFlight/inav-configurator/pull/2025), but not sure.

Using Speedybee F405 WING running Inav 7.1.0 - Airplane platform type, Flying wing mixer preset

[INAV_7.1.0_cli_20240427_104402.txt](https://github.com/iNavFlight/inav-configurator/files/15143516/INAV_7.1.0_cli_20240427_104402.txt)

My board is mounted in an orientation that is 90 degrees positive YAW rotation from facing forward.  This was setup in the Alignment tool.  The Setup page and telemetry all react properly as the plane is moved about various axis, showing proper position with pitch, roll, and yaw.  When in Angle or Acro modes (the only two I have currently setup besides manual) the rotation of the plane does not apply the proper control surface movement to correct the position to level flight.  It appears to be oriented to the position of the flight controller if it was mounted facing forward in the default configuration.

Is there something I am missing in setting this up?  I assume that shifting the orientation of the board would also affect the stabilization controls and properly adjust the ailerons to compensate.

One additional thought - If you do change the alignment of the flight controller, should you then run a calibration of the accelerometer in the new configuration?  If that is expected, I did NOT do that yet.  Now that I think of it, it makes sense to do that (perhaps??)  Again - the orientation in iNav and telemetry is correct without this.  I calibrated with the normal orientation of the flight controller.  It is the other flight modes that don't appear to move the control surfaces correctly.  Based on the note in Accelerometer Calibration - 

> Note: If the IMU is mounted in another angle or on the underside of the flight controller. Do the calibration steps with the IMU pointing as shown in the pictures, not the quad (otherwise calibration won´t work.

It appears doing the calibration of the accelerometer should be done in the "normal" orientations to the flight controller.

Thank you.

sensei-hacker, 2024-04-28:
The accelerometer calibration is only calibrating the accelerometer, not the aircraft. It's recommended to do that before installing the board in any aircraft. It is not affected by how the FC is later installed in various aircraft.

The issue you're experiencing is very likely one tow things. Either channel mapping, or a mixer issue.  The sign is likely backwards on two of the mixer lines.

To confirm, first go the Receiver tab and move one stick at a time, a long one axis at time, and ensure yaw goes up when you move the yaw stick stick right, roll goes up when you move the roll stick right, and pitch goes up when you move the pitch stick forward.

Next, the mixer. Mostly likely, two of the mixer lines have the wrong sign (positive vs negative). To start, ignore the right elevon and look only at the left one. Move the pitch only. Does the left elevon move down slightly when you push the pitch forward?  If not, reverse the sign on the pitch line for the left elevon.  Now roll on the left elevon only - does it rise when you command left roll?  If not, reverse the sign.

Repeat for the right elevon.

If that all checks out and it still looks like they aren't moving correctly for stabilization, we'll go back to the setup page. The 3D model might look like it's moving correctly - but be upside-down or backwards.  With the model pointed away from you, into the screen, the wing marked in red should be on the LEFT side.  The green wing is the right wing.

You are also welcome to join the INAV Discord if you'd like to walk through these steps with someone.

Jetrell, 2024-04-29:
@jbrandwi If its related to PR 2025. The easiest way to prove it. Is to turn off the magnetometer.. This will prove if the IMU and Mag are _slightly_ clashing.
A magnetometer isn't a essential for fixedwing navigation. But it does improve heading estimation in 7.1.

RomanLut, 2024-04-29:
Currently there are known problem with alignment tool in 7.1:
1) It works properly for horizontal FC orientation only (either normal or upside down). Vertical or tilted orientation does not work  ( It should work correctly for your setup )
2. It works for horisontal MAG orientation only (it will NOT work for your setup).
3. With PR 2025 ( which is not included in 7.1), any FC orientation will work correctly. 
4. I am not 100% sure mag orientation works corectly with PR2025 until I test it.

So I would suggest to:
1) disable MAG temporary.
2) Set FC orientation using Alignment tool.
3) Check that plane is moving propertly in the Setup tab.

If plane is moving properly in the Setup tab, but control surfaces are moving incorrectly, then there is problem with mixer and RC channels mapping/AETR/inversion.

4) once all work correclty, you may try to enable MAG and try to find correct orientation by trial and error, so the plane still moves correctly in Setup tab. However MAG is usefull for Cruise only. I would leave MAG disabled in your setup.

jbrandwi, 2024-04-29:
Closing due to misconfiguration.

closed 2024-04-29 by jbrandwi