iNavFlight/inav-configurator #2058: [MASTER] Unreliable connection

breadoven, 2024-05-03:
Current master seems to be unreliable when switching between tabs and when saving changes. e.g. :
Advanced Tuning tab ... change WP turning smoothing setting ... click Save and Reboot.
Reboots to Status screen but screen gets stuck loading ... you have to disconnect and connect again.
On reconnecting the Status Screen works most of the time but can hang randomly.
When you can connect WP turning smoothing setting in Advanced Tuning tab is unchanged ... save failed.
The same happens for other settings in Advanced Tuning.

Console output when this happens indicates Settings saved in EEPROM:

```
Getting setting nav_land_minalt_vspd
F:\INAVDev\inav-configurator\js\msp\MSPHelper.js:3078 Getting setting nav_emerg_landing_speed
F:\INAVDev\inav-configurator\js\intervals.js:114 Killing all intervals except: global_data_refresh,msp-load-update,ltm-connection-check
F:\INAVDev\inav-configurator\js\msp\MSPHelper.js:1373 Setting set
F:\INAVDev\inav-configurator\js\msp\MSPHelper.js:640 Settings Saved in EEPROM
F:\INAVDev\inav-configurator\js\msp\MSPHelper.js:724 Reboot request accepted
F:\INAVDev\inav-configurator\js\intervals.js:114 Killing all intervals except: global_data_refresh,msp-load-update
F:\INAVDev\inav-configurator\js\intervals.js:114 Killing all intervals except: global_data_refresh,msp-load-
```
Changing setting in CLI works OK.
Selecting other tabs sometimes needs a couple of tries because they also get stuck loading.

This is for FLYWOOF411 target on Windows 10.

rts18, 2024-05-04:
@breadoven Same results here.

DzikuVx, 2024-05-05:
Yup, there are issues with `master` . Several, I suppose we will save them one by one

DzikuVx, 2024-05-05:
@breadoven might be fixed by #2063

DzikuVx, 2024-05-05:
but it's not done yet for some tabs, will dig deeper

DzikuVx, 2024-05-07:
This looks like issue with MSP load on slower boards. But we won't be supporting F411 in INAV 8 so probably priority of it is slightly lower now

Jetrell, 2024-05-10:
Has anyone else noticed the OSD tab crashes if an ALT page is selected. Or if the save button is pressed in the OSD tab ?
Windows build with F405 FC.

breadoven, 2024-05-10:
> Has anyone else noticed the OSD tab crashes if an ALT page is selected. Or if the save button is pressed in the OSD tab ? Windows build with F405 FC.

I don't get a crash selecting ALT pages but I do if I try and Save. (F722 windows)

DzikuVx, 2024-05-10:
Guys, if something does not work make a new ticket. It's hard to track comments in different tickets

DzikuVx, 2024-05-10:
#2065

DzikuVx, 2024-05-17:
I think tab switching issues should be resolved with #2082

closed 2024-05-17 by DzikuVx