iNavFlight/inav-configurator #2061: GPS Model is not aligned wth North direction.

mmosca, 2024-05-05:
These pictures were taken in the alignment tool, with the gps in the same orientation (CW 270 FLIP) If the Arrow on the GPS is supposed to point North, it does not match the axis view (X axis is yellow and is supposed to point North.

North Arrow pointing to the back of the quad:
[image]

X Axis pointing to left side of the quad:
[image]

This is probably one of the sources of confusion in the alignment tool.

mmosca, 2024-05-05:
@RomanLut any chance you can fix the generic gps model for 7.1.1? I see you have a PR to add specific models for many of the common gps units in 8.0. :)

mmosca, 2024-05-05:
For some reasno I pasted 2 of the same images. But the axis represent the result of the gps model with the arrow pointing to the same orientation. Isn't the X axis of the Magnetometer the North direction, according to the sensor?

That is the source of my confusion.

mmosca, 2024-05-05:
I updated the model on the original entry.

The arrow should point to the reference North direction (X or the magnetometer) axis on the gps model.

RomanLut, 2024-05-05:
No it does not. The only meaning of an arrow on magnethometer is "If you mount magnetometer with arrow pointing to front, then set CW270flip preset".

Axes represent magnetometer local frame. Actually Matek-m8q has axis markings:

[image]

RomanLut, 2024-05-06:
Ok, the confusion seems to be the assumption that arrow on Magnetometer model points to the North. Actually Magnethometer model represents Matek magnetometers (M8Q, M9N, M10Q) as most widely used models. Arrow on Matek magnetometers points to the negative Y axis. 
https://github.com/iNavFlight/inav-configurator/pull/2056 should resolve it.

mmosca, 2024-05-06:
> No it should not. The only meaning of an arrow on Matek magnetometer is "If you mount magnetometer with arrow pointing to the front, then set CW270flip preset".

 I guess this is the issue. This makes no sense (other than historical reasons). There is also plenty of gps units with no arrows, or different axis orientations, as you probably found out when modeling a bunch of them. 

The 270 in CW270flip is the offset of the X axis from the axis tat points to the front of the quad.

closed 2025-01-18 by b14ckyy