iNavFlight/inav-configurator #2085: [Master] Tuning tab hangs using F405 controllers

Jetrell, 2024-05-19:
When using the configurator master build to date. 
And using the master firmware post [2719472](https://github.com/iNavFlight/inav/commit/2719472787a269d7e443e47301ce4b878ef7f031) . I can not get any F405 controller I have tested, to access the Tuning tab. It just hangs indefinitely.

But after speaking with @breadoven who said he had not had this issue. But also referenced he was using an F722.
I then proceeded to try the above builds with one of my F722 controllers.. And after a slight delay, the Tuning tab loaded.
So it appears this issue is surround the F405 controllers.

For reference. I tested.
`Mamba F405`
`Matek F405`
`Speedybee F405`
`Omnibus F405`

Windows 64 OS

breadoven, 2024-05-19:
I tried this with an F411 and an old Omnibus F4 Pro (F405 I believe) and the Tuning tab works OK for both. That's with the latest Configurator master and fairly recent INAV master (2 weeks old perhaps).

Don't see why #9471 would affect the Tuning tab though. I'm not aware any of the changes are used for that tab at the moment.

Jetrell, 2024-05-19:
@breadoven I'm not totally sure of what the go is either. But these are the results I get. 

And why I can use the one of the last nw.js versions of the configurator on all the firmware's of have tried with success.
While the previously mentioned results are what I see with electron.

I tested everything I could think of before opening this issue report. Because I was second guessing myself, when not hearing of any issue from anyone else.  But I don't know what else it could be ?

breadoven, 2024-05-19:
Do you run Configurator in development mode so you can get error info from the console ? May not show anything though if it just hangs.

closed 2024-06-17 by Jetrell