iNavFlight/inav-configurator #2090: iNav Configurator V 7.1.1 5 inch quad wizard freezes

AlmousaM, 2024-05-20:
**Flight Controller: SpeedyBee F405 V4
iNav Configurator Version: 7.1.1
iNav Firmware: 7.1.1**

Upon connecting the flight controller for the first time to the iNav configurator and selecting the 5 inch quad quick wizard, the wizard popup the saving window and it stays on the screen forever. i tried reseting / refreshing with full chip erase but the issue was still there.

 I downgraded to iNav 7.1.0 and the wizard works fine without any issue, so i think the problem is with iNav configurator 7.1.1 !

eLEcTRiCZiTy, 2024-05-21:
Can confirm for the 3" quad. Erased the flash, the clean flashing of INAV 7.1.1, picked a defaut settings, and a small windows with a message "Device - Saving default settings" forever.

terryfritz99, 2024-05-22:
I am having the same problem with all presets on a Speedybee F405Mini and a Matek F405HDTE. If you close that window by clicking on the X there are no motors configured.

snaewe, 2024-06-24:
Duplicate of #2100 ?

closed 2024-06-28 by DzikuVx