iNavFlight/inav-configurator #2097: No motor output detected in output tab

gabrielvachon, 2024-05-24:
I have a 5" quad with a Speedybee F7 V3 flight stack on which i flashed INAV 7.1.1. Everything seems to work fine, however, the outputs tab show no motor slider (only master) for testing and the ESC doesn't seem to receive any signal from the FC because there is no second sound after I plug the battery. I then tried to flash INAV 6.1.0 and the motors are displayed and the ESC makes a sound to tell that it receives a signal.

[image]

Step to reproduce : 
- Flash a SpeedyBee F7 V3 stack with INAV 7.1.1
- Set the model to 5"
- Go to the outputs tab

In conclusion, i can use INAV 6.1.0 but not INAV 7.1.1 on my FC. Is there a known way to fix this?

KrackJack1, 2024-05-25:
i got the same issue but solved now in mixer tab press load and apply on quad X
hope this help.thx

[image]

gabrielvachon, 2024-05-25:
It worked for me, thanks a lot!

closed 2024-05-25 by gabrielvachon