iNavFlight/inav-configurator #2099: Mission Control tab not working

voxtelnismo, 2024-05-27:
How to replicate:
Open INAV configurator. Connect cable to FC. Select proper port in configurator. Connect to FC. Once the main configurator panel comes up, click on Mission Control tab.

Result: Blank window with "Waiting for Data" icon.

Tried with: Matek F411 Wing, Kakute F7 AIO, Matek F405 Wing.
Operating systems: Windows 11, MacOS, Ubuntu 22.04, Ubuntu 24.04, Kali 2024.01.

breadoven, 2024-05-28:
Which INAV version ?

voxtelnismo, 2024-05-28:
> Which INAV version ?

Updated original post. I tested it with 7.00 flight control and 7.00 and 7.11 configurator.

Works fine in Speedybee Bluetooth app.

breadoven, 2024-05-28:
You need to use Configurator 7.0.0, same as the firmware.

voxtelnismo, 2024-05-28:
> You need to use Configurator 7.0.0, same as the firmware.

Solved. Thank you.

closed 2024-05-29 by breadoven