iNavFlight/inav-configurator #2100: Default parameters saving freezing on first visit

ZipovUA, 2024-05-27:
Hi all.

I tried SpeedyBee F4 V3, SpeedyBee F405 Wing.
Often have situation 
When I install firmware (7.1.1)
AND connect to INav Configurator
AND set 7 inch mini quad (or tailed plane in case for SpeedyBee F405 Wing)
THEN I see that pop-up (that informs that params are saving) is freezing and we can not say: did saving process finished or not

ZipovUA, 2024-05-27:
[image]

ZipovUA, 2024-05-27:
[image]

ZipovUA, 2024-05-29:
In fact I see that my problem is duplicate for https://github.com/iNavFlight/inav-configurator/issues/2098

So here are steps:
1. Flash frimware on speedybee f405 v3 or speedybee wing
2. Connect to inav-configurator
3. Select pre-set (default settings) for corresponding flight model (in my case it's 7 inch quad or v-tail plane)
4. Wait until settings will be saved (in fact we see that saving settings pop-up is never gone)
5. In about 10 minutes close pop-up
6. Go to mixer page
7. Now we see that pre-set was not saved

McLogo76, 2024-05-29:
> In fact I see that my problem is duplicate for #2098
>
> So here are steps:
>
> 1. Flash frimware on speedybee f405 v3 or speedybee wing
> 2. Connect to inav-configurator
> 3. Select pre-set (default [...]

Btw. I went back to configurator 7.1 just to be sure and the issue was then gone, so it has to be the 7.1.1 configurator.

snaewe, 2024-06-21:
After applying the selected preset, the FC is set to "PID Profile 2" and then gets stuck (as seen on the image above).  I think you could just sitch to PID profile 1 and then get the correct settings (PIDs etc.).

McLogo76, 2024-06-21:
> After applying the selected preset, the FC is set to "PID Profile 2" and then gets stuck (as seen on the image above). I think you could just sitch to PID profile 1 and then get the correct settings (PIDs etc.).

Yes, that's what I did. But it should not switch to PID Profile 2 imo.

mszulen, 2024-06-27:
Same here on newest configurator after flashing 7.1.2 to SpeedyBee F405 wing (both mini and regular) on Mac M1

closed 2024-06-28 by DzikuVx

mszulen, 2024-06-28:
Paweł, dziękuję! :)