iNavFlight/inav-configurator #2107: Motor unresponsive after upgrade 6 to 7.1.2

flyinginger, 2024-06-09:
Reproduce:
flashing inav 7.1.2 and clean in configuration onto flywoo F745nano. All config ok but motor #2 won't move anymore
Expected
All motor work and move on to fine tuning
Fix , work around
Reinstall inav 6.1.1

sensei-hacker, 2024-06-13:
Can we see a screenshot of your mixer, please?

mmosca, 2024-08-19:
Stale issue.

closed 2024-08-19 by mmosca

windelwilli, 2025-05-19:
Would be great if u would share the solution, got the same problem ;-)

sensei-hacker, 2025-05-19:
> Would be great if u would share the solution, got the same problem ;-)

Can we see a screenshot of your mixer, please?

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
Why isn't the servo 3 green and why is there a warning on the Motor tab ?

sensei-hacker, 2025-05-20:
You have the motor set to -100% throttle.  Negative.
You want the motor to rev up for braking, and the motor to stop to make the plane go?  That should probably be 1 not -1.

I see you have a rule for the third servo, and a rule for the fourth servo.  What happened to the first servo and the second?  You probably want servo #1 and servo #2 if you have only have two servos. You can't have a fourth servo without having a first and second. :)

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
Servo 1 and 2 didnt solve ist... 

And other combination look like this

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
[image]

windelwilli, 2025-05-20:
Is the green and red importand to work correctly ? Because if i choose 2 and 3 then physically it do wat it have to do... bitte then there isnt the green one like in the first picture

sensei-hacker, 2025-05-20:
> Is the green and red importand to work correctly ? Because if i choose 2 and 3 then physically it do wat it have to do... bitte then there isnt the green one like in the first picture

Green and red is not important, if it does what it needs to do.

It looks like the numbering got a bit confused. To make the numbering and colors correct, it _MAY_  help to delete both servos, save and reboot, then add them again.

windelwilli, 2025-05-21:
The reeboot and add the servos again hit it.... now everything is green and red and everything is okay.... have done..... thx