iNavFlight/inav-configurator #2109: [SpeedyBee wing] INav configurator does not come to UDF mode, when ELRS (CRSF) is installed

ZipovUA, 2024-06-13:
Environment:
 inav-configurator version - 6.1+
 Flight controller - speedyBee F405 wing (have no problem with F4V3)

Steps:
1. Take some speedyBee wing with ELRS that's installed into CRSF port.
2. Click on boot button, connect by USB to computer and open inav-configurator
3. Take a look at available ports in inav-configurator - we see that inav configurator recognizes connection as USB connection
4.  disconnect USB wire
5. take off ELRS from Speedybee wing device
6. connect by USB wire again with holding boot button

Now we can see that inav-configurator set in ports list DFU port

I don't know, maybe there is some special reason to have such behaviour, but will be great if will be no need to take off ELRS from speedy bee wing before flashing software and put it again after firmware was installed and making first visit to system :)

sensei-hacker, 2024-06-13:
F4 (and F7) MCUs can be flashed over UART1 or UART3.
So when you try to flash it, if something is talking on UART1 or UART3 it will try to get the firmware from that port.

For this reason, it's traditional to put your receiver and GPS on UART2 and UART 6.  If you have something on UART1 or UART3, power it off or disconnect it while flashing.

andrewnewton, 2024-06-30:
Try disconnecting GPS

mmosca, 2024-08-19:
And make sure your radio is off as well.

This is a well know issue with STM32 mcus, and the work arround is to disconnect / turn off devices connected to UART1 and UART3.

GPS is a common offender.

closed 2024-08-19 by mmosca