iNavFlight/inav-configurator #2110: [MASTER] Advanced setting, OSD and Mission Control won't load, MSP error

breadoven, 2024-06-15:
Not possible to load Advanced Settings, OSD and Mission Control tabs with current master connected to INAV current master firmware.

Throws up the following MSP error which I'm pretty sure is nothing to do with anything I've done. Also not sure when it become a problem exactly but must be a recent issue:

[image]

breadoven, 2024-06-15:
It happens every time on the tabs mentioned above for an F722 board configured as a multirotor. However, testing on an F411 board configured as a fixed wing it only happens on the OSD tab, although this is using a slightly out of date version of the current master.

JulioCesarMatias, 2024-06-15:
This is also happening to me. I just reported this on the Dev Team.

JulioCesarMatias, 2024-06-15:
I managed to reproduce the problem you described, but when clearing the configurator cache, everything worked normally.

[image]

breadoven, 2024-06-15:
Ah well done that fixes it. Is that cache reset button a recent addition ?

Thanks !

mmosca, 2024-06-15:
> Ah well done that fixes it. Is that cache reset button a recent addition ?
> 
> Thanks !
> 
> Can't help but think this is going to catch people out who won't know to click that button.

Yes. It is Pawel doing some magic to speed up the configurator. Shouldn't be a problem on release code, but changing back an forth between different branches or older snapshots of master can cause issues if you don't reset the cache

stronnag, 2024-06-15:
It would be neat of the cache included the build id (commit hash) and if the cached build id didn't the current id, then automagically clear the cache. Would save a load of support issues for "nightly" builds.

breadoven, 2024-06-15:
I added something to flush the cache on disconnect ... but maybe that somewhat defeats its usefulness.

mmosca, 2024-06-15:
A better approach would probably be to fetch missing item on cache miss.

rts18, 2024-06-16:
I attempted to run builds of both masters from yesterday.
These are the results.
The OSD, Outputs, Failsafe, Tuning, Advanced, MissionControl and GPS tabs hang.

[image]

And worse. The configuration tab does not load any sensors.
I set them to auto and saved. Nothing changed.
[image]

The configurator was run on a computer with Windows 10 OS.

>However, testing on an F411 board configured as a fixed wing it only happens on the OSD tab.

I have noticed the same. Every FC target I tested worked differently. Some fail to load a few tabs. And others fail to load a dozen tabs. 
I attempted to clear the cache, with no change here.

DzikuVx, 2024-06-16:
The issue may appear when setting descriptors changed. In general, once final release is out, there should be no problems as then settings will not change. For now, in case of any problems, flush the cache 
@breadoven cache flush on disconnect completely breaks the idea
@stronnag the idea of build id if probably the best.

PR #2111 (2024-06-17): Extend settings cache with build info
Fixes #2110
closed 2024-06-17 by DzikuVx