iNavFlight/inav-configurator #2130: Regarding substitutes for Bing Map after discontinuing service

wskkys, 2024-07-01:
[Bing Maps for Enterprise](https://learn.microsoft.com/bingmaps)  is deprecated and will be retired. Enterprise account customers can continue to use Bing Maps for Enterprise services until June 30th, 2028. Existing free (Basic) account customers can continue to use Bing Maps for Enterprise services until June 30th, 2025.
Can the author consider replacing the Bing map with another service provider？

DzikuVx, 2024-07-02:
That depends which. Google Maps are not a possibility as UAV tracking is against Google Maps TOS

JonnyRemix, 2024-08-14:
Hi, Bing maps is deprecated and is replaced by Azure maps. Is it possible to use Azure maps as a substitute? Thanks

CarolineTyler, 2024-08-17:
Yes, I got the notice also for the bing maps API being depreciated. We really could do with an alternative.

cptX, 2024-08-19:
Are the Bing Maps Api for Inav 7 already deprecated or is it going to be depricated? If the second then when exactly? I need to use it for the next 2 weeks but my account is locked and I need to give a mobile phone to unlock it, which I don't usually do very happily. Please tell me if it is already deprecated I don't need to do it...

mhk1058, 2024-08-20:
> Are the Bing Maps Api for Inav 7 already deprecated or is it going to be depricated? If the second then when exactly? I need to use it for the next 2 weeks but my account is locked and I need to [...]

If you already have an account you are OK until June ‘25.  If you don’t have a maps account it won’t accept  new users or allow new API creation.

sensei-hacker, 2024-08-21:
Where.com seems to be the likely alternative. That's where Bing sources some of their satellite images. One can get a free API key.
Someone will need to do a bit of coding to make it work.

OptimusTi, 2024-08-21:
What's up with openstreetmaps.org?

stronnag, 2024-09-02:
Or  [Mapbox](https://www.mapbox.com/) via `mapproxy`.
[image]
(requires free Mapbox API key).

mhk1058, 2024-09-02:
> Or [Mapbox](https://www.mapbox.com/) via `mapproxy`. ![Screenshot from 2024-09-02 16-36-04](https://private-user-images.githubusercontent.com/158229/363747067-575a30db-83e9-449e-ba61-669a2a8af21e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzMTI3NTksIm5iZiI6MTcyNTMxMjQ1OSwicGF0aCI6Ii8xNTgyMjkvMzYzNzQ3MDY3LTU3NWEzMGRiLTgzZTktNDQ5ZS1iYTYxLTY2OWEyYThhZjIxZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwMlQyMTI3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTI0ZGNlYjY3ODYzZGQzMmU2ZDAwZTM4MmY0Y2VmOTE2NGYxZGU2YjA2OGQ3YzMyODMyMTI4MGU0NTE5OTI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2NuBJR97m402JejUH8c8rJD7H7wh-6E_94o4XYOI7qs) (requires free Mapbox API key).

Can you explain how you get that to work? I have an API key but no idea how to set up the map proxy URL

stronnag, 2024-09-03:
> Can you explain how you get that to work? I have an API key but no idea how to set up the map proxy URL

You need a locally accessible instance of `mapproxy`, i.e. you host `mapproxy` on your LAN.
The configurator repo provides information [MAPPROXY.md](https://github.com/iNavFlight/inav-configurator/blob/master/MAPPROXY.md) on using `mapproxy`. However, as it starts by recommending the installation of an obsolete, 10 year old version of Ubuntu, one might doubt its providence. 

My contemporaneous set of [notes](https://gist.github.com/stronnag/74d94f9784d647b74f71e8e32ec6be54) details modern usage and installation. These notes includes examples for MapBox and other useful services like OpenTopo.

mhk1058, 2024-09-03:
> You need a locally accessible instance of `mapproxy`, i.e. you host `mapproxy` on your LAN. The configurator repo provides information [...]

Many thanks... probably beyond my skill and patience, I'll live with open street map for now :-)

mhk1058, 2024-09-03:
> What's up with openstreetmaps.org?

Nothing really, especially if it is sufficient for your needs. However, satellite photographic maps can give you an easier to read picture of the type of terrain and topo maps (like Ordnance Survey in the UK) give more intuitive terrain contouring.

stronnag, 2024-09-30:
A few more practical considerations:

* There  has been some developer discussion as to whether to maintain the extant "OpenLayers" API or  to migrate to "Leaflet"; until a clear consensus / direction emerges, developing Bing alternatives may be nugatory.
* Both OpenLayers and Leaflet can support Azure Maps and MapBox
* For either (or both of) Azure and MapBox, volunteer developer effort is needed and users will need to obtain personal access key(s).

b14ckyy, 2025-01-18:
FYI: work on additional map providers is in process. No final decision yet but we will very likely add ESRI Maps as a free Satellite image provider with further commercial solutions added as well. 

More info will follow in time when we get closer to 8.1 (or whatever comes before June)

daniel1302, 2025-04-15:
Here is a MapProxy config for Google Maps, but please note that it is for educational purposes only, you cannot use Google maps tiles via proxy :)

```

services:
  demo:
  wms:
  tms:

layers:
  - name: opentopo
    title: OpenTopo
    sources: [mapbox]
  - name: gm
    title: OpenTopo
    sources: [gm_cache_satelite]

caches:
  opentopo:
    sources: [opentopo]
    grids: [webmercator]
    cache:
      type: file
      directory_layout: tms

  gm_cache_satelite:
    grids: [webmercator]
    sources: [gm]
    cache:
      type: file
      directory_layout: tms

sources:
  gm:
    type: tile
    grid: GLOBAL_WEBMERCATOR
    url: http://mt0.google.com/vt/lyrs=y@0&hl=en&x=%(x)s&y=%(y)s&z=%(z)s

  opentopo:
    type: tile
    grid: GLOBAL_WEBMERCATOR
    url: https://a.tile.opentopomap.org/%(z)s/%(x)s/%(y)s.png

grids:
    webmercator:
        base: GLOBAL_WEBMERCATOR

globals:
  cache:
    base_dir: ./mapproxy_cache
```

But if you want to use mapproxy(ofc not for proxing Google Maps) you can follow the following steps:

NOTE: Sometimes instead of `python` in your terminal you may use `py`

1. Download python: https://www.python.org/downloads/windows/
2. Save mapproxy config(simillar to the above) in the file `C:/mapproxy.yaml`
3. Open powershell and enter the following commands:
   - [Install python package manager](https://pip.pypa.io/en/stable/installation/): `py -m ensurepip --upgrade`
   - Install mapproxy: `pip install MapProxy==2.2.0`
4. Start mapproxy in powershell ` mapproxy-util serve-develop C:\mapproxy.yaml`

Then in INAV you would set the following values:

[image]

- Map provider: MapProxy
- Map API key: empty
- MapProxy URL: http://localhost:8080/service?service=wms
- MapProxy Layer: gm

Scavanger, 2025-06-11:
#2385