iNavFlight/inav-configurator #2140: Mistakes in configuration

cptX, 2024-07-07:
1. In INAV Configurator 7.1.1 in Failsafe the failsafe_throttle is displayed in uS units which is wrong. This should be a value without units showing the absolute value of the throttle channel.
[image]

2. failsafe_off_delay is not clear if it affects the RTH Procedure or only the Land Procedure. If it affects only the land procedure the separation line should be under the failsafe_off_delay field not above. Also, because this is not clear, I don't know If I leave the value to 200ds if the multicopter will drop during RTH Procedure if more than 20s are passed. Please clarify this!!

3. Also I would like to ask you to provide a way to customize all failsafe parameters through gui not only cli.

MrD-RC, 2024-07-07:
1. Nope, that is correct. It states uS and is displayed in uS. Just like other throttle values in Configurator and the CLI. 

2. Failsafe delay affects failsafe. If RTH is activated during, it will use the RTH settings and not kill the motors. 

3. Configurator is for the most needed parameters. The other CLI parameters are not needed for most builds and have not been included in the UI for this reason.

cptX, 2024-07-08:
Hi @MrD-RC, regarding 1. I guess you are right. I suppose these uS represent the duration of the "servo" pulse.
Regarding 2 I really believe this separation line should be under the **failsafe_off_delay field**. In general it is very unclear if this setting will affect the RTH. I have tried to find info online and I couldn't so I had set this to 0 to disable it. Without proper clarification who would risc a motor turn off during RTH. Please consider adding more clarification in the infos.
Can you please clarify my 3. bullet too? Probably when you responded you didn't see my updated questions which had been 1 to 4. Thank you in advance.

MrD-RC, 2024-07-08:
3. Not really needed. I don't know anyone who has actually changed those values. 

4. No, for the reasons stated above. Only the common, safe parameters should be on the Failsafe page. The advanced parameters, which could cause problems should remain CLI only.

Jetrell, 2024-07-09:
@cptX 
`failsafe_off_delay` is the case of a copter, it is primarily use is for the failsafe LAND procedure throttle time out..
But its still applicable in RTH emergency landing fall-back. If the absolute worse case occurred from microwave jamming. Causing the RX to failsafe, and the GNSS module to loose satellite fix.. And IF your FC doesn't have a barometer.
Under this condition,`failsafe_off_delay` and `failsafe_throttle` will be used to land the copter at the location the issue occurred. And other than that, these setting will have no effect on normal navigation RTH. 

You mentioned your multicopter. But are asking about fixedwing settings.

`failsafe_fw_roll_angle` 
`failsafe_fw_pitch_angle` 
`failsafe_fw_yaw_rate `

When a RX failsafe occurs. With `nav_rth_allow_landing = ALWAYS or FS`. The airplane will roll, pitch or yaw according to these settings.. They should be set to allow the plane to perform a gentle descending spiral landing.
Meaning those values become the new control surface center positions (like if the user had applied trim offset). With stabilization work around those values.
Defaults are generally best. Because setting them too high if its windy. May send the airplane into a spiral crash.
_Those setting do not apply when using the new 7.1 fw auto land feature._

This may help too - https://github.com/iNavFlight/inav/wiki/GPS-Failsafe-and-Glitch-Protection#emergency-landing

sensei-hacker, 2024-07-09:
> have tried to find info online and I couldn't so I had set this to 0 to disable it. Without proper clarification who would risc a motor turn off during RTH.

FYI consider you come across anything that seems unclear to you, where there two interpretations:

Interpretation A: The default settings are insane
Interpretation B: The default settings are reasonable

The correct interpretation is pretty much always B. Defaults were selected and reviewed by multiple knowledgeable people, and flown successfully by many thousands of people.

This can be useful to keep in mind if you're considering setting anything ten times higher than the default or things like that.

closed 2024-11-15 by b14ckyy