iNavFlight/inav-configurator #2141: MATEK F405 v1

MBR595, 2024-07-07:
Hello,

I recently changed controllers. Currently using a FRSKY TARANIS X-LITE PRO with the TBS module and receiver. I'm using a Matke F405 V1 board and INAV Configurator 7.1.1. FW is 7.1.0 MATEKF450SE. My RX is wired to UART4 but it's not being recognized in the configurator when I turn it on, save and reboot. Any advice you can offer is greatly appreciated.  

[image]

sensei-hacker, 2024-07-08:
You wrote "my rx is wired to UART4".
Your screenshot shows it's configured for rx on UART2.

MBR595, 2024-07-08:
Hello,

It may be connected to UART 4. The writing on the board is very small. Assuming it's connected to UART 2, it's still not recognizing the RX. Any suggestions?

sensei-hacker, 2024-07-08:
In your shoes, the first thing I would do is use some good lighting and take a close-up picture with my phone. Then zoom in on that pic to a) verify which user it is on and b) check for any stray wire strands, solder whiskers, etc.

If it's not connected to the same uart that is configured, that would be the problem. So I would start there.