iNavFlight/inav-configurator #2143: speedybee f7v3 in inav 7.1.2  motor mapping not working

samsab1, 2024-07-11:
Hello everyone
I'm running into an extremely annoying problem whose cause I can't determine:
inav 7.1.1, speedybee f7v3 in inav 7.1.2
Flight Controller reversed 180°, engine remapping done correctly I believe, dshoot 300.
almost every time I connect my controller card to INAV, I lose the motor mapping; more precisely, the mapping remains faithful to what I had determined BUT when starting the drone, only 3 motors turn, or only 1 (the one which was not previously turning when the other 3 were turning) or even only the front 2 turn or then only the 2 rears rotate..... in short almost all combinations are possible, while this same flight controller card, with inav 7.1.1 firmware, oriented normally (not inverted therefore) and the standard mapping, does not didn't make that kind of bad joke to me.

Can anyone tell me how to fix this permanently?
I would nevertheless point out that sometimes the 4 motors start working when the drone is armed.....

samsab1, 2024-07-11:
I add pictures
[image]
[image]

samsab1, 2024-07-11:
i would say:
with and without prop
batterie connected
throttle : using the inav interface above

sensei-hacker, 2024-07-11:
Sorry, I didn't see the pictures you had added. The Outputs tab is indeed the best place to test.

I said you "remapped" the motors. Can you tell me please how you went about doing that?

samsab1, 2024-07-12:
I mounted my drone in a classic way respecting the standard assembly of the F7, that is to say arrow of the FC towards the front of the drone.
then I wanted to rearrange this assembly and so I turned the complete stack, already wired with its motors, by 180°. finding that the order of the motors was incorrect, I remapped the FC with the inav utility.

each time I modify another parameter of inav such as the RTH parameters, I take the risk of seeing my motors not all start the next time the motors are armed. I don't understand what can bring about this random aspect, although very present; that being said, as I write, the 4 engines are recognized and functional, but for how long?

under inav 7.1.1 in normal configuration, from the orientation point of view of the fc, I have never had this problem

aveias100, 2024-08-25:
Hi samsab1!
I have the same problem... there is a way to resolve the problem?

sensei-hacker, 2024-08-25:
Again, what do you mean you "remapped" the motors?  There is no such thing in INAV. So what exactly did you do?

aveias100, 2024-08-25:
> I just re-read and noticed this:
> That sounds like a dodgy connection on the plug between the FC and ESC.

Motors are in wrong position in inav. It was ok and working in betaflight..
Changing to inav motor mapping is wrong...

aretak, 2024-10-01:
Im having same issue with microairf405v2 on 7.1.2 when checking ESC configurator only two of the four pop up initially I noticed when going through the mixer and checking proper rotation that motor one and two Spin but not three and four. I checked for pins being smashed in the connector, etc. Which is what led me to the configurator however, when I change absolutely nothing else using the same connector I plug-in a separate flight controller and all four ESC show up.

sensei-hacker, 2024-10-01:
> Im having same issue with microairf405v2 on 7.1.2 when checking ESC configurator only two of the four pop up initially I noticed when going through the mixer and checking proper rotation that motor [...]

Can we see a screenshot of your Mixer tab, please?

aretak, 2024-10-01:
[image]

aretak, 2024-10-01:
ok weird so i added a 5th motor and everything works 
[image]

sensei-hacker, 2024-10-02:
In your first screenshot you put a _servo_ on S3.

aretak, 2024-10-02:
thats the thing im just loading mixer not adjusting

sensei-hacker, 2024-10-02:
Yep, if you load up some servos and some motors in your mixer, you would need to look at where they are.

If you want them on different outputs, you can change them.

Of course INAV can't peer into your our mind and put them where you want them, without being told where you want them.

closed 2024-10-02 by sensei-hacker