iNavFlight/inav-configurator #2147: Random movement in quad position when GPS is enabled

cptX, 2024-07-16:
When being in the configurator in the Setup Tab I was viewing random movement of the quad, although it was sitting still on the table. Please see the attached video.

https://github.com/user-attachments/assets/f6efd37e-6abd-4a24-a194-5c9cb5d2fa80

Initially I thought that a sensor was giving noise but I checked all inputs at the Sensors Tab and everything was stable. Then I disabled the GPS and the random movement of the quad in the Setup Tab was gone.

This made me very curious why the system was making random movements on roll and pitch axis if the gps gives info only about the position in the horizontal plane and provides absolutely no info of the roll and pitch orientation? Is this a bug of the inav firmware, a bug of the inav configurator or no bug at all? If the gps info affects pitch and roll in firmware this could lead to unstable flight and worse position hold.

Jetrell, 2024-07-17:
@cptX That's normal when GNSS position data is being used to update/augment/correct the attitude and heading reference.. The effect is worse when the delusion of precision is higher and the FC is not to actively stabilizing the aircraft, as it would be in-flight.

cptX, 2024-07-18:
@Jetrell I don't understand why getting new position data from gps affects the stationary pitch and roll. What I mean, is what we see here the result for the fc kalman filters to change the pitch and roll accordingly in order to move to the new position the gps has given? GPS cannot give info about pitch and roll so how the pitch and roll angles are affected if all sensors are providing the same input?

Jetrell, 2024-07-18:
>GPS cannot give info about pitch and roll so how the pitch and roll angles are affected if all sensors are providing the same input?

Axis position is based on gyro rotation and the accelerometer confirming that by measured gravity vector. But this has its limitations when forces other than 1G are applied in flight.. It can lead to the accelerometer reference drifting.

There are dynamic accelerometer weights, using rate ignore, slope and turn. To help correct the error when the aircraft experiences forces greater or less than 1G for a short period of time.. But outside of that, GPS VELNED is used to compensate for speed and centrifugal forces applied to the accelerometer.

Both these methods are combine. And the smoothness in which they transition will vary, depending on how _poor_ the satellite position accuracy is.. This is the jittering you see when the aircraft is motionless with a 3D Sat fix.

You can read up more about how it works in the code if you wish.

Nimalankarunakaran, 2024-08-05:
Hi,i have the same problem please let me know the solution if you have solved the problem thanks

sensei-hacker, 2024-08-30:
> Hey, i have the same problem and my fpv crashed twice because of this, im using inav 7.1. 2 with f405 v4 speedybee and M10Q-5883 gps/compass. Any one help please how to reduce this bad behavior at least.

It is unlikely that your aircraft crashed while sitting on the desk because the gyro senses vibrations from a truck driving by on a nearby street. If you are having some problem with your blimp, rover, quadcopter, airplane, or submarine which resulted in a crash, please a open discussion or issue about your problem, and fill in the form so that we can have the information necessary to maybe help you.

zied923, 2024-08-30:
Idk how to open a discussion here, and sorry if my question wasn't in the right place. I sow the same topic and i asked, that's it.

zied923, 2024-08-30:
I m new here so idk how to use github, i will open discussion ( just did it) thank you.

closed 2024-11-15 by b14ckyy