iNavFlight/inav-configurator #2148: [INAV 8.0] GPS tab does not always update

mmosca, 2024-07-19:
When entering the GPS tab in configurator, the packet count/update rate don't always update.

Often fixed by switching to different tab and back.

Seen on Macos  and Windows.

b14ckyy, 2025-01-19:
have you seen that again recently? I think I have had this months ago as well but not seen in a long time since RC Phase.

mmosca, 2025-01-23:
I think the gps might be fixed. I will keep an eye on it a bit longer and close the issue.

I think there are still some state gremilins in configurator, specially when you are flashing / making big changes, but I may open a separate issue to track these.

b14ckyy, 2025-01-23:
The only thing I see more frequently again is the empty "Status" page after connect that needs a second or even a third try. Especially though slow connections like mLRS. But I think that was a different issue.

sensei-hacker, 2025-01-25:
> It appears that you managed to install iNav 8 on MacOS. Where to get Latest Configurator? There is none for MacOS on release page((

The first four files are all Mac.

[image]

Tumerlan, 2025-01-25:
> > It appears that you managed to install iNav 8 on MacOS. Where to get Latest Configurator? There is none for MacOS on release page((
> 
> The first four files are all Mac.
> 
> [image]

Thank you so much! I dig all google to find it.
It is not so obvious, cause Inav 7.1.2 had file called macOS. iNav 8 has Darwin instead.