iNavFlight/inav-configurator #2150: [INAV 8.0] Sensor status is not always updated

mmosca, 2024-07-19:
When connecting / disconnecting to a fc, the sensor status bar on top does not always update.

The GPS sensor or the magnetometer may not turn blue, for example.

MrD-RC, 2024-11-07:
There seems to be a few issues with the states not being collected properly. I have also noticed that the failsafe parachute (https://github.com/iNavFlight/inav-configurator/issues/2232) and modes auto channel detect (https://github.com/iNavFlight/inav-configurator/issues/2233) no longer work too. I have a feeling that this is all related to the move to electron?

0crap, 2024-12-03:
Seen that as well with 8.0 RC1 on the Sensors tab, when monitoring the Barometer.
The green line is moving (Baro) but the blue line (Alt) is not.
Alt is stuck at zero.

Flashing target (MATEKF405) with 7.1.2 and using Configurator 7.1.2 shows the blue line (Alt) moving around as normal.

MrD-RC, 2024-12-03:
I believe this is fixed in RC2.

0crap, 2024-12-03:
Awesome!

closed 2024-12-05 by b14ckyy

0crap, 2024-12-07:
> I believe this is fixed in RC2.

I tested 8.0 RC2 and the Baro Alt is not updating for me.

I started testing with the FC on 7.1.2 and at that point the Baro Alt is updating correctly on Configurator 8.0 RC2.
As soon as I update the FC to 8.0 RC2 it's not updating anymore.
So this makes me believe it could be a FC firmware issue?
Let me know if you want me to open a new GitHub issue on the FC part.

A minor Configurator improvement could be to set the default scale to 100 on the Gyro part.
For me the default value of 2000 shows a flatline, when set to 100 it looks much better.
(But maybe this is only an issue with the FC I'm using, which is a MATEKF405)

[image]

b14ckyy, 2024-12-07:
Different topic. Please open a new Issue ticket with your information. I think I have a similar issue but never checked the sensor page again. Will add to that.

0crap, 2024-12-07:
Will do!
(The Gyro Scale to 100 default remains a minor Configurator issue.)

b14ckyy, 2024-12-07:
Oh btw it could be intended that the Alt is not updating when disarmed. The raw gyro altitude below is what actually matters i think.

0crap, 2024-12-07:
Done.
https://github.com/iNavFlight/inav/issues/10510

0crap, 2024-12-07:
> Oh btw it could be intended that the Alt is not updating when disarmed. The raw gyro altitude below is what actually matters i think.

It all works and updates fine with v7.1.2, also on the bench disarmed.