iNavFlight/inav-configurator #2152: Stuck on "Saving default settings"

pranas, 2024-07-20:
I was using iNav Configurator 7.1.1 (on macOS 14.5) to flash MATEKF411TE with 7.1.2 (with full chip erase). On the first connect, I select "Airplane with a Tail" and the configurator gets stuck on "Saving default settings". I can re-flash or click "Reset settings" and the issue is there as well. I can close the message and continue configuration but how can I be sure that all default settings were applied fully?

[image]

pranas, 2024-07-20:
Reverting back to iNav Configurator 7.1.0 fixed the problem so it seams specific to 7.1.1.

sensei-hacker, 2024-07-20:
Yeah that's a known issue with 7.1.1.
You can close it after 20 seconds, or use 7.1.0

closed 2024-07-20 by sensei-hacker