iNavFlight/inav-configurator #2153: iNAV 7.1.1 - After VTOL 4+1 Config Mixer preset displays motor numbers out of order

packetkiller, 2024-07-22:
After successful VTOL 4+1 configuration, when in profile 2 (quad profile) pid profile 2, the Mixer preset image motor numbers are not in correct display order. Same can be said for when in profile 1 (FW mode) pid profile 1, the plane shows all motors on the front vs just the one pusher/puller.

Hardware:
Matek H743-WINGv3
4 standard quad motors
1 standard pusher motor
airspeed sensor
GPS
opticflow sensor
rangefinder sensor

FC FW: iNAV 7.1.2
Using iNAV Configurator 7.1.1

diffs, dumps available by request
Supplied screenshots of MC mode, transition mode, and FW mode showing mixer screen.
Happy to supply any additional information needed or that I may have missed.
I do not know if this affects operation. I suspect it is just an iNAV Configurator display thing, and thus if this can be confirmed, we may be better served by making this a feature request.
[image]
[image]
[image]

sensei-hacker, 2024-07-23:
> for when in profile 1 (FW mode) pid profile 1, the plane shows all motors on the front vs just the one pusher/puller.

The preview accurately shows that all of those motors are set for throttle only - they are not set to induce roll or yaw.

sensei-hacker, 2024-07-23:
But for practical use - when you have a motor turned off, it doesn't matter where it is. :)

packetkiller, 2024-07-23:
If the first motor (S1) is a pusher, that would be a throttle motor right? I would think that would show as a S1 in the back of the plane on the FW mode image. The 2nd motor is a dummy (not really there, nothing connected), so can skip using the second timer 3 entry. The 3rd to 6th motors (S3 to S6) are quad motors facing up. If I understand what you are saying, those should be set differently by adding a "1"s in the roll, pitch, and yaw columns. I thought that is what I had but did I miss something? When the motor is set to "off" with the "-1" in the Throttle column, do I also need to still have the other columns set? Appreciate the guidance.

packetkiller, 2024-07-23:
Per other discord discussions, will use a -2 to disable or turn off a motor that is not used for throttle in a profile (dummy motor, pusher when VTOL in MC mode). Since the images are just attempting to depict/represent where the motors are, and do not affect the mixer operation, will ignore.
Suggested next steps:
1) update documentation to clarify VTOL mixer settings for "off" motors and how to configure "dummy" motors to achieve desired timer and FC motor control slots (S1, S2, etc).
2) iNAV 7.1.2 / iNAV 7.1.1 Configurator - 4+1 VTOL configuration, add -2 for "off" motors or to disable dummy motors
3) Suggest a feature request to update the iNAV 7.1.1 Configurator mixer screen to handle custom motor and servo configurations better by adding a specific VTOL image(s) and differentiate between pusher/puller and up facing motors
4) Suggest a feature request to update the iNAV 7.1.1 Configurator mixer screen to handle the difference between pusher and puller motors (use a boolian) to get the image more representative to reality, even if it not affect the inner workings of the FC.

closed 2024-07-23 by packetkiller

kritigyanischal, 2025-09-15:
@packetkiller i urgently need ur help. Ping me on instagram or discord or telegram pr even whatsapp or email. @x2envy is my instagram and nischalkritigya@gmail.com is my email
 Pls as soon as possible.