iNavFlight/inav-configurator #2162: Missing MSP Option for the Receiver wizard

b14ckyy, 2024-08-08:
When i set up a new craft and using the new Setup Wizard, I have no option to select MSP as the main RC source. 
This would be needed for MSP-RC from mLRS (And maybe, just maybe in the future for ELRS as well).
[image]

mmosca, 2024-08-19:
That is because this assumes serial rx. And MSP is not an option for serial rx.

```
# 
# get receiver_type 
receiver_type = SERIAL
Allowed values: NONE, SERIAL, MSP, SIM (SITL)

# get serialrx_provider 
serialrx_provider = CRSF
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, IBUS, JETIEXBUS, CRSF, FPORT, SBUS_FAST, FPORT2, SRXL2, GHST, MAVLINK, FBUS, SBUS2
```

b14ckyy, 2024-08-20:
You think it makes sense to add the Receiver Type to the Wizard? Not sure how many people will set it to MSP or SIM in the future. Its also confusing that MAVLINK is a serial Rx provider and not a receiver type here. as it also needs to be SerialRx turned off in the Ports tab.

mmosca, 2024-08-20:
The main possible driver for this would be mlrs or elrs mavlink, in which case we need to add more than just msp. But I still think they will be somewhat much options and most users will use one of the serial rx options.

b14ckyy, 2024-08-20:
I am fine with one or the other. 

Either add receiver_type to the Wizard or put MSP as a Serialrx_provider. 

Maybe moving MAVLINK to receiver_type though might make more sense as all the serialrx_provider options need the Serial Rx switch in Ports tba to be on. MSP and Mavlink do not. This might be a more streamlined option. 

additionally there is a chance that even ELRS might get MSP-RC support at some point as I know from 2 major ELRS devs that they like the idea.

mmosca, 2025-01-21:
I will put it as tentative for 9.

Since mavlink can do "serial rx" without being a serial rx, it is likely possible to move msp rx to the same list as the other rx protocols.