iNavFlight/inav-configurator #2178: Add Programming Tab support for vtx pit mode

MartinHugh, 2024-08-25:
Currently it is possible to use the Programming tab to control vtx Power, Band and Channel.

It would be really useful to add the ability to set vtx Pit Mode as well, so that this can be set (for example) from a transmitter switch.

MartinHugh, 2024-08-26:
I note that for HDZero (which is my use case) this functionality is possible by setting the VTX power to max index +1, which configures 0mW.  It is possible to switch from 0mW through all the power levels.

I am not sure if there is any demand for this feature on system other than HDZero, but since it works for that system using the above workaround, I am happy for this issue to be closed.

b14ckyy, 2025-01-18:
One Last note: 
PIT mode is not implemented in INAV due to limited SmartAudio functions on many VTX where PIT mode can cause issues. Maybe in the future.

closed 2025-01-18 by b14ckyy