iNavFlight/inav-configurator #2179: Random roll and pitch after enabling gps

zied923, 2024-08-30:
This issue is discussed before but i don't see any solution so im opening this topic again maybe somene figured out how to solve it. 
Im on inav 7.1. 2
Speedybee f405 v4
Gps M10Q-5883
I have 9/10/11 sat lock (indoors) 

Uploading VID_20240830_180104.mp4…

[image]
[image]

The issue is roll and pitch become not stable after enabling gps, drone drift badly and crash with gps enabled. Without gps all is good.
I have lidar installed the tf luna and it hover almost perfect with surface / angle mode. No any issue. 
Wire are isolated and gps mounted in the arm to keep it far of interference. 
Campass showing the right north no issues. 
What could be wrong? Any ideas? 

( i joined pictures and small videos)

zied923, 2024-08-30:
https://github.com/user-attachments/assets/dbb23e06-cd58-4868-a0c2-14b00c187600

zied923, 2024-08-30:
[image]

zied923, 2024-08-30:
[image]

zied923, 2024-08-30:
[image]

zied923, 2024-08-30:
[image]

zied923, 2024-08-30:
https://github.com/user-attachments/assets/a919b2d0-ffec-4172-80b1-cad13b98d929

sensei-hacker, 2024-08-30:
Your video shows roll deviation of less than 0.1 degrees. Turning upside down would be 180°, so what your seeing is less than 1/1000th of that.

Similarly for pitch - half a degree. That will have absolutely no effect on flight whatsoever. Just compensating for a light breeze is several degrees.

If you are having some flight trouble, please provide information about that. What flight mode, precisely what you saw occur, such as "right right tendency upon throtl

closed 2024-08-30 by sensei-hacker

zied923, 2024-08-30:
Why you are having bed attitude with me??.? I told you im new with github and im new with fpv ,and i don't know how to use github properly, and What u think it is? Its a quadcopter, i mentioned all details , doesn't matter i will fix it by myself.

sensei-hacker, 2024-08-30:
> What u think it is? 

We have absolutely no idea what it is, for as long as you refuse to tell us. As long as you refuse to fill out the form or tell us anything about the vehicle or the flight, we have no way of having any idea what problem you might be having.

I asked you before, in the other issue you hijacked, whether it's a submarine, a blimp, a wall crawler, a quadcopter, a tailless plane ...
Unfortunately, we can't read your mind, so until you tell us about the problem, we can't fix it for you.

I'm trying really hard to help you, but of course I can't without any information.

zied923, 2024-08-30:
The issue is abviously so clear but your making of it a big deal! With your rules or whatever, what template i have to fill? I don't see any.. Im using a phone, i told you im new IDK HOW TO USE GITHUB, and i put all the info, but really doesn't matter, your making a simple question a big deal, i know you know what im talking about, its just i know the kind of person are you, the kind of ppl who complicate everything and making of it a big deal.. Your waisting your time and mine with nonsense, have a nice day and last time i will post here and i will leave github also.

sensei-hacker, 2024-08-30:
> The issue is abviously so clear but your making of it a big deal! 

What you've told us so far is that something crashed. That's the info we have so far.

What happened before the crash?
Did it pitch forward?  Did the motors cut off?  Did it run full speed into the side of a house?

What flight mode were you in?

What kind of craft are we talking about?

OptimusTi, 2024-08-30:
wow...

Jetrell, 2024-08-30:
@zied923 There is a possibility this is related to a known issue. As referenced [here](https://github.com/iNavFlight/inav/pull/10243).
I'm not totally convinced this hasn't crept in as a side effect of work added to 7.1, even though much of that testing in that link, was done with the master build of INAV 8.0.

Regardless, the more information you can add, the better.
Things like a 1) Log 2) Diff settings file 3) Was this flight outside ? 4) How many tests did you conduct ?  And as @sensei-hacker said, video footage is always helpful too.. We need to be sure it is related to the above. And not just caused by poor position accuracy (2.5m isn't great). You did mention something about being indoors.
Something I would suggest. Is for you to load a build of 7.0 on your quad. And report back if it has the same issue.. And also providing a log for that test as well would help.

zied923, 2024-08-31:
Thank you for looking at this issue. 
From the tests i made ( im new but im not stupid im an electrical engineer) , i think there is something not stable with inav 7.1.2, why in betaflight the gps lock better and faster then  inav, with the same hardware and the same setup, just plug the quad and satellite jump to 15, with inav it took more time and less sattelite... I will make more tests to pruve it and collect more data to share it. 
All my test are  indoor ( i can't outdoor). But 8 or 9 sat locked it should be enough. No? 
My HDOP is between 1.5 and 2.5. 
why the gps make the pitch and roll behaving randomly even with drone set still in the desk..i mean the others sensors tell that the drone is not moving why the gps try correct a position while the other sensors say that the position is not changing. 
Is  there any filters to minimize this effect? I'm not a programmer but i have little knowledge about programming, i think its possible but first the pbm have to be isolated and latter try to fix it. 
About my quad is 5 " with f405 v4 fc gps is M10Q-5883. 
My Gps is working perfectly because i try it with betaflight and all is good. 
My compass is also good it showing the correct direction without any interference. 
My test are all made in Angel mode. I didn't try other mode because it will behave worse.
I will try today load inav 7 and see how it behave and let you know. 
PS : LIDAR interfere wity gps, i installed the TF luna and just plug the battery the number of stattelite drop to 0, i changed the lidar to a UART far of gps and the gps worked normally lile before ( i wanted to share this with ppl who want install lidar to avoid this pbm) 
Sorry again if this thread is bothering some ppl, as i said I'm new and i don't know how to use Github or how to fill  the correct template.