iNavFlight/inav-configurator #2185: Feature Request: Create a VTOL preset with the defaults from the documentation

Pairan, 2024-09-14:
VTOL Keeps banging in the heads of many pilots and as INAV is capable of it, it would come in handy to have a preset to select after flashing the FC.

b14ckyy, 2024-11-15:
I thought about this as well. Generally I like that idea. 
But probably not for the initial presets as there are too many possible VTOL configurations. 

Technically you select the craft type that should be the Hover mode, so a multirotor. Then you switch to mixer profile 2 and load a Fixed Wing mixer. 

So potentially we can add extra Multirotor mixes but I am not sure how applicable it would be to also set settings like the transition and everything. But we will talk about this.

Pairan, 2025-01-13:
>But probably not for the initial presets as there are too many possible VTOL configurations.
Sad but true...

Maybe an option would become to make a tab like the Multirotor / Tricopter / Airplane Tab ... e.g.

Tricopter > VTOL Fixed wing
Multirotor > VTOL Fixed wing
...

Adding the usual transition mode, MAX and linked profiles