iNavFlight/inav-configurator #2203: Linux/BLE: no devices, no errors.

BortEngineerDude, 2024-10-02:
I am running iNav configurator v7.12 on a Linux laptop and I cannot use BLE mode to connect to a quad with SpeedyBee F405 V4 FC; meaning the BLE menu is empty. iNav log and stdout seemingly have no errors at all, log files attached. Toggling "Wireless mode" changes nothing. FC is discoverable in Bluetooth settings and in the bluetoothctl CLI tool, but cannot be paired. However, using SpeedyBee app on an Android smartphone or using USB cable works just fine.
Any ideas what went wrong?
[image]
[bluetoothctl.log](https://github.com/user-attachments/files/17232753/bluetoothctl.log)
[inav-configurator.log](https://github.com/user-attachments/files/17232754/inav-configurator.log)

stronnag, 2024-10-02:
BLE does not work in INAV 7.* configurators. 
* It will "sort of" work in the IANV 8 configurator
* You can use a BLE  <> TCP/IP bridge if it really bothers you.

stronnag, 2024-10-02:
FYI duplicate of https://github.com/iNavFlight/inav-configurator/issues/2091

closed 2024-10-02 by stronnag