iNavFlight/inav-configurator #2212: Configuration TAB not open after build!!! only the configuration tab!!!???

jmcdrone, 2024-10-17:
I build myself a .hex file from omnibusF4V3  on visual code without any error. But on the inav configurator i could not open the configuration tab. All other tabs are normally working!! Is that a bug? Best Regards

jmcdrone, 2024-10-17:
> Which version number did you build, and what did you change?

I use the code from github and run the command make OMNIBUSF4V3 on visual
code. I get a .hex file and flash to my omnibusf4 flight controller. I
change the softserial from s6 to a led pad. It works but i have to use cli
to activate softserial feature because the configuration tab doesn't open
on inav configurator

jmcdrone, 2024-10-17:
I build the newest version

MrD-RC, 2024-10-17:
Are you using the configurator from master? It will not work with the 7.1.2 release configurator.