iNavFlight/inav-configurator #2220: MATEK H743 WING V1 : ACCELEROMETER ICM 20602 NOT OFFERED

samsab1, 2024-10-26:
### Discussed in https://github.com/iNavFlight/inav-configurator/discussions/2219

<div type='discussions-op-text'>

<sup>Originally posted by **samsab1** October 26, 2024</sup>
hello, i try to convert this FC to INAV 7.1.2 but it doesn't work: the list of accelerometer does'nt match what it is required: i mean ICM20602. when running betaflight this FC works, but uses ICM20602
[image]

how can use this FC with INAV? Is it possible to get a firmware for that FC for last INAV?
best regards from France
</div>

mmosca, 2024-10-26:
ICM20602 uses the MPU6500 driver in INAV.

Set you gyro to auto and let inav auto detect it.

Otherwise, post a picture of your fc.

closed 2024-10-26 by mmosca

samsab1, 2024-10-27:
[image]
Setting gyro to auto does not give any result

samsab1, 2024-10-27:
[image]

mmosca, 2024-10-28:
@mateksys any idea what is going on here?